nextcloud/nextcloudpi #1: Optionally Disable HTTPS

cathalgarvey, 2017-03-19:
Hey, awesome work and thank you sincerely! I've been planning to put NextCloud on a Pi3 for ages, but I wanted to do a docker-job, and docker on ARM is slightly painful.

The only thing about this I don't love is: I use [caddy](http://caddyserver.com/) to host my site, and I'd rather not open another port and set another DNS entry for Nextcloud. I'd like instead to add a rule to caddy (which runs on another Pi) to proxy to the NextCloud Pi over the LAN, so there's only one DNS rule and open port to Caddy (which handles HTTPS transparently), and Caddy handles proxying to Apache.

With auto-HTTPS and HTTP disabled, this is slightly a pain, also. I strongly believe Lets-Encrypt powered HTTPS should be the default.. but could there be an optional config change to disable HTTPS if desired?

cathalgarvey, 2017-03-20:
> Hello,
> Well, you are the second person that comes with this request, so I will
> add it to `nextcoludpi-config` soon

Thank you kindly! :)

nachoparker, 2017-03-24:
implemented

closed 2017-03-24 by nachoparker

blissarts, 2020-09-18:
Hello,
I also want to avoid the https as I do it over a reverse proxy.
However I cannot find the option to disable the forced ssl connection. Nextcloud always forwards me to the https scheme instead of staying on the pure http.

Can you please point me in the right direction?

Thanks, BR

ovpc, 2020-09-18:
Ask questions on  [forum](https://help.nextcloud.com/)  and link to it here for future reference, if you please, thank you.