nextcloud/nextcloudpi #2: no-ip installer "Installation to ... failed"

jonwhittlestone, 2017-03-29:
Hi

I am attempting to run the installer for no-ip on my raspberry pi B running raspbian jesse and see the message 

    Launching installation
    Installation to 192.168.0.24 failed

I suspect this is not the solution, but I am trying this whilst sshed in remotely via the noip domain in question. Would this be an issue?

Thanks for your time

----

**What I've already done**

run `sudo apt-get update` and `sudo apt-get upgrade`

I have installed `dialog` and run the following

    ./installer.sh no-ip.sh 192.168.0.24

I see the dialog that says _Enter the desired configuration for no-ip_

I enter my noip credentials

    Active: yes
    user: <email>
    pass: <pass>
    time: 30

I press **<Start>**
    
**The error message**

    Launching installation
    Installation to 192.168.0.24 failed

**The remaining thing to try**

I suspect this is not the solution, but I am trying this whilst sshed in remotely via the noip domain in question. Would this be an issue?

nachoparker, 2017-03-31:
can you specify what version of the installer are you using?

`git describe --tags --always`

Just to be clear: installer.sh is **NOT** meant to be run inside the raspberry pi, but from your computer.

If 192.168.0.24 is the IP of your rpi, then you must issue the command 

    ./installer.sh no-ip.sh 192.168.0.24

from another PC on your local network.

If you changed the default user or password for the raspberry pi, you have to specify it on the command line. 

    PIUSER=nacho PIPASS=ownyourbits ./installer.sh no-ip.sh 192.168.0.24

see 

https://ownyourbits.com/2017/03/05/dynamic-dns-for-raspbian-with-no-ip-org-installer/

jonwhittlestone, 2017-04-01:
ok, thanks. I have found another solution now anyway (built into my ISP's router)

closed 2017-04-01 by jonwhittlestone