nextcloud/nextcloudpi #4: Enable ssh on first boot

Plorenzo, 2017-04-02:
I would like to try the image but I don't have any display or keyboard so I need ssh enable form the beginning to be able to use the Pi.

With the normal Raspbian image you can achieve this adding a file called ssh to the SD root (it doesn't matter the content).

I tried this with NextCloudPi but it's not working.

nachoparker, 2017-04-02:
This image is based in Raspbian, so there is no difference

If you place a file called ssh it indeed enables SSH

If you look at 

https://github.com/nachoparker/qemu-raspbian-network/blob/master/qemu-pi.sh

The relevant code is

```
# prepare the image
SECTOR1=$( fdisk -l $IMG | grep FAT32 | awk '{ print $2 }' )
SECTOR2=$( fdisk -l $IMG | grep Linux | awk '{ print $2 }' )
OFFSET1=$(( SECTOR1 * 512 ))
OFFSET2=$(( SECTOR2 * 512 ))

mkdir -p tmpmnt
mount $IMG -o offset=$OFFSET1 tmpmnt
touch tmpmnt/ssh   # this enables ssh
umount tmpmnt
```
I use that everyday (automatically) to build NextCloudPi images, so I know it works.

You must have done something wrong... maybe you forgot to sync and umount?

Plorenzo, 2017-04-02:
maybe I'm missing something but I unmount the SD (on my Mac) after I run:
`sudo dd bs=1m if=NextCloudPi_03-31-17_FULL.img of=/dev/rdisk3`
and I add the ssh file 

After I insert the SD to the pi, attach the ethernet to the router, I find the IP the Pi has been assigned and I try 
`ssh pi@192.168.1.12`
the result is: 
`ssh: connect to host 192.168.1.73 port 22: Operation timed out`

nachoparker, 2017-04-02:
how are you adding the ssh file to the boot partition?

did you run sync before dd'ing the image to the SD?

Just to be sure:
1) mount image
2) touch bootmnt/ssh
3) sync
4) dd to SD card

Plorenzo, 2017-04-02:
I'm not sure what you mean by sync, I add the file with touch from the boot directory. 

This are the steps I'm doing:
1. Download image from https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/ 
2. Insert SD into mac
3. `diskutil unmountDisk /dev/disk3`
4. `cd ~/Downloads`
5. `sudo dd bs=1m if=NextCloudPi_03-31-17_FULL.img of=/dev/rdisk3`
6. `cd /Volumes/boot`
7. `touch ssh`
8. Eject SD
9. Insert SD into RPi
10. `ssh pi@192.168.1.12`

And then I get one of the following error: 
```
ssh: connect to host 192.168.1.73 port 22: Operation timed out
ssh: connect to host 192.168.1.73 port 22: Host is down
```

nachoparker, 2017-04-02:
The steps look ok to me, but I do not own an apple computer, so I am not sure.

In any case, it must be just the same as with any other Raspbian. The exact same

Connect a TV and keyboard if you are stuck

Plorenzo, 2017-04-02:
After some time trying to figure it out I think it has to do with DHCP no assigning a local IP to the Pi .
Any idea why it could be?
If I do the exact same steps with the raspbian image, works fine

nachoparker, 2017-04-02:
I tested the issue and you are right:

as long as wicd is active, DHCP no longer appears to work.

I will disable it by default then

thanks for the feedback

nachoparker, 2017-04-02:
opened as a new issue #5

closed 2017-04-02 by nachoparker

Plorenzo, 2017-04-02:
Thank you for the contribution and the quick answers! I'm looking forward to try NextCloudPi :)