nextcloud/nextcloudpi #7: Error nextcloudpi-config change database directory

egofont, 2017-04-21:
I am using an external hard drive to store my information and the Nextcloud database as well as the swap file as it fills the micro sd.

I'm using a Raspberry Pi 3.

After starting Raspberry for the first time, I give it nextcloudpi-config, there is a new update, I update the version of Nextcloud.

I setup the external drive in the fstab
I mount the drive in the / media / external directory.
In the / media / external folder I created a folder for the database and another folder for the Nextcloud files.

I use the nextcloudpi-config command and I selected to change the database folder but it generates a mysql error, I tried stopping the mysql service before running nextcloudpi-config and it also fails:

Currently the error I have is:

root@raspberrypi:/# systemctl status mysql.service
● mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql)
   Active: failed (Result: exit-code) since Fri 2017-04-21 01:25:36 UTC; 6s ago
  Process: 1559 ExecStop=/etc/init.d/mysql stop (code=exited, status=0/SUCCESS)
  Process: 1640 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)
 
Apr 21 01:25:32 raspberrypi mysqld[1804]: 170421  1:25:32 [Note] InnoDB: Highest supported file format is Barracuda.
Apr 21 01:25:36 raspberrypi /etc/init.d/mysql[2093]: 0 processes alive and '/usr/bin/mysqladmin --defaults-file=/etc/mysql/debian.cnf ping' resulted in
Apr 21 01:25:36 raspberrypi /etc/init.d/mysql[2093]: [61B blob data]
Apr 21 01:25:36 raspberrypi /etc/init.d/mysql[2093]: error: 'Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2 "No such file or directory")'
Apr 21 01:25:36 raspberrypi /etc/init.d/mysql[2093]: Check that mysqld is running and that the socket: '/var/run/mysqld/mysqld.sock' exists!
Apr 21 01:25:36 raspberrypi /etc/init.d/mysql[2093]:
Apr 21 01:25:36 raspberrypi mysql[1640]: Starting MariaDB database server: mysqld . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . failed!
Apr 21 01:25:36 raspberrypi systemd[1]: mysql.service: control process exited, code=exited status=1
Apr 21 01:25:36 raspberrypi systemd[1]: Failed to start LSB: Start and stop the mysql database server daemon.
Apr 21 01:25:36 raspberrypi systemd[1]: Unit mysql.service entered failed state.

What is the procedure that I must follow to be able to change the folders of the database and the files?

Thank you

nachoparker, 2017-04-21:
I will add to the conversation that your external drive filesystem is NTFS, as you mentioned on the blog comments.

Is that correct?

In my past experience, mounting NTFS and FAT systems is prone to problems because they do not support permissions.

Also I did have a NTFS mounted on my Raspberry Pi 1 permanently in the past and I had to reformat it to EXT4 because the performance was just terrible.

In any case, could you post the output of `mount` and `ls -l /media` `ls -l /media/external`

Thank you for the feedback

Thanks for the feedback

egofont, 2017-04-21:
Thank you. This is my line in fstab.

/dev/sda1       /media/externo  auto nofail,uid=33,gid=33,umask=0027,dmask=0027,noatime 0 0

nachoparker, 2017-04-21:
ok, so you are giving the external drive permissions for the `www-data` user, but not for the database user `mysql`

at the very least that is one problem that you have. You could try umask=0000 (dmask,fmask) but that would not be ideal from the point of view of security. 

Also note that the NTFS driver can take all your CPU easily, at least it used to a couple years ago.

nachoparker, 2017-04-26:
I will assume that this was a configuration issue that you had. Please, reopen or open a new issue if you find other problems.

closed 2017-04-26 by nachoparker