nextcloud/nextcloudpi #12: Problems with moving data and database directories

noether, 2017-05-20:
Raspberry 2 B v1.1 and Image NextCloudPi 11.0.3.

I have mounted with fstab 'uuid=xx /media/Nextcloud defaults 0 2' a USB disk.

When I run for the first time the config tool for moving the data directory to /media/Nextcloud/data, the script exists with a green comment (seems correct). However, although I can still communicate with Nextcloud from my desktop I have the following error:

Data directory (/media/Nextcloud/data) is invalid
Please check that the data directory contains a file ".ocdata" in its root.
Cannot create "data" directory (/media/Nextcloud/data)
This can usually be fixed by <a href="https://docs.nextcloud.com/server/11/go.php?to=admin-dir_permissions" target="_blank" rel="noreferrer">giving the webserver write access to the root directory</a>.

If I try to move again the datadirectory, the script fails (it says the same error). I can confirm that the data directory is successfully created and the .ocdata exists (the directory seems correct).

If I move the database directory to /media/Nextcloud/database, I can see that the dir is created and there are files there (again they seem correct). However, I lose communication with Nextcloud

Error 500: Internal server error.

The error from the script suggest to run some commands:

‘Job for mysql.service failed. See ‘systemctl status mysql.service’ and ‘journalctl -xn’ for details.’

systemctl status mysql.service
● mysql.service – LSB: Start and stop the mysql database server daemon
Loaded: loaded (/etc/init.d/mysql)
Active: failed (Result: exit-code) since Sat 2017-05-20 14:47:10 UTC; 5min ago
Process: 1551 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)

journalctl -xn
No journal files were found.

nachoparker, 2017-05-20:
Thanks for the feedback

What is the filesystem of the USB stick?

Have you copied everything? If so, I will remove the long technical questions from the blog comments

noether, 2017-05-20:
The filesystem is ext4. It is a fresh disk, so far it looks good.
Yes, I have copied all the relevant info from the blog :P. Thanks!

noether, 2017-05-20:
I solved the first issue of moving the data directory by assigning 777 rights to /media/Nextcloud, i.e., to the whole device (not only to the data dir). Mmm, it seems it is a permissions problem, what group/user shall I add to the /media/Nextcloud ?

nachoparker, 2017-05-20:
The requisite when you mount the folder is that the user `www-data` has read-write-access access to it

Check the permissions for the user `www-data` to the files inside `/media/Nextcloud/data/

Try something like

````
sudo -u www-data ls       /media/Nextcloud/data/.ocdata/
sudo -u www-data touch /media/Nextcloud/data/.ocdata/
````

noether, 2017-05-20:
that solved the problem. It seems that www-data needs permissions to the device (not only the folder in the device). I have also found this in the installation manual of NextCloud, so in the end, it was my fault :P. It would be nice to advise this permissions issue in your conf tool or blog :P.

Thanks for your time !

nachoparker, 2017-05-20:
great

As an improvement, I'll add a check for that before moving the datafolder or database.

I am about to release the automount feature, so that will also make this easier for users.

thanks!

closed 2017-05-20 by nachoparker

wayneoutthere, 2018-11-07:
so... question related to this (trying to set up with encrypted drive and get both database and datadir over there...)

is there an order in which this process should occur?  My current plan (and tutorial that I'm writing) is as follows:

1. mount encrypted drive to /media/USBdrive
(give this drive correct permissions??  Not sure what these would be either...)
2. create with mkdir /ncdata and /ncdatabase
3. give them correct permissions (still working on that one)
4. go into NCP admin and move datadir (this worked at least gave me green light)
5. move database (this has been a big fail for a long time now)

Thanks!
i'm wondering if perhaps I'm missing a critical step or have the order wrong..

nachoparker, 2018-11-08:
yes, it seems like nc-datadir didn't move the files correctly, you can check the contents from the terminal