nextcloud/nextcloudpi #13: Can't upload files >~100 kB

Jakethethird, 2017-05-25:
Installed on RPi2 interface opens fine. Can download test files.

- When only on local network after initial installation, I could upload a 1.1 MB file. 
- Once letsencrypt, fail2ban and  and noip set up, I can't upload anything larger than 100 kB, either when logged on inside network (192.168...), or from outside via URL.
- Message is consistently "Request Entity Too Large" on Nextcloud interface (screenshot included).
- Full reinstall, reconfiguration...same result.
- Default max file upload size is 768 MB for both admin and new user.

Thanks.

[image]

nachoparker, 2017-05-25:
Is it possible that you have modsecurity enabled?

Jakethethird, 2017-05-25:
Yes! I've just turned modsecurity off, rebooted and now I can upload larger files without issue. My NC just upgraded to v.0.12.4, but this problem still existed after that...until I turned of modsecurity.
You say modsecurity is experimental so do you recommend just turning it off, or changing the config? Without it, but just with fail2ban, is it still pretty secure when open to the web? 
Presumably fail2ban is protecting the nextcloud login (long random string as password) and not the pi and username of raspbian?

Thanks. Brilliant project!

nachoparker, 2017-05-25:
Great.

1) you do not need to reboot after changing any config unless otherwise stated :)
2) sorry, if you had it already active updating to >0.12.1 is not enough, but you had to reactivate from nextcloudpi-config

fail2ban only protects: SSH and brute force login attempts (too many failed attempts to log in)

if you have your security updates in place and a recent version of NC that should be pretty good. 

Modsecurity though... is a BEAST. It is the most secure thing in the world, but it takes a lot to configure it. It works for me with the nextcloud apps I use, but if another user installs another app chances are modsecurity will block it by default and things won't work.

That is why I warn that is experimental... if it does not work for you, you will have to disable it.

Hopefully I will get people's feedback (and/or help) to write a nice ruleset for Modsecurity with NC.

I suggest you read my posts on them

https://ownyourbits.com/2017/03/23/modsecurity-web-application-firewall-for-nextcloud/
https://ownyourbits.com/2017/02/24/nextcloudpi-fail2ban-installer/

closed 2017-05-25 by nachoparker