nextcloud/nextcloudpi #14: Nextcloud 12 permissions issue on external USB

clanger7, 2017-05-26:
Hello and thanks for creating this!
I used your 12.0.0 file and it installed fine, I changed default pw and set for my region etc, I also connected an ext4 external HDD (sda1) and used the nc-automount option – I didnt do anything with fstab as it was not in instructions that I saw…
I moved the db and data to the USB drive and I see them there with www-data and mysql owners
pi@pi3:~$ ls /media/USBdrive -all
total 32
drwxr-xr-x 5 root root 4096 May 26 02:22 .
drwxr-xr-x 3 root root 4096 May 26 02:20 ..
drwx—— 2 root root 16384 May 26 01:51 lost+found
drwxr-x— 8 www-data www-data 4096 May 26 02:40 ncdata
drwxr-xr-x 5 mysql mysql 4096 May 26 03:00 ncdatabase
I was able to connect and login to the web GUI and create a user for myself, and also connect via phone
BUT I get permission denied listing the data directory AND I cannot create folders or files as my user I created in the web GUI or on my phone app, it just says “error creating folder”

pi@pi3:~$ ls /media/USBdrive/ncdata -all
ls: cannot open directory /media/USBdrive/ncdata: Permission denied

I saw another user claim this same issue and he provided a link to edit /etc/fstab – so i did that and now my permissions are thus:
pi@pi3:~$ ls /media/USBdrive -all
total 32
drwxrwxr-x+ 5 pi pi 4096 May 26 02:22 .
drwxr-xr-x 3 root root 4096 May 26 02:20 ..
drwxrwxr-x+ 2 pi pi 16384 May 26 01:51 lost+found
drwxrwxr-x+ 8 pi pi 4096 May 26 02:40 ncdata
drwxrwxr-x+ 5 pi pi 4096 May 26 03:33 ncdatabase

But now the Web GUI will not load at all and I still cannot create folders from the phone app.

So what permissions are needed or what steps are needed for a clean install of your 12.0.0 with a USB ext4 drive already attached? the nc-automount seemed to say this would do all of this for you but I’m hitting these permission issues.

Is there a way to easily reverse the changes I made or should I start from scratch and reflash the sd card image? If so , what is the proper step to ensure correct permissions so I can actually create folders and upload files?

Thanks much in advance! 🙂

clanger7, 2017-05-27:
**Original setup**

burn 12 version to SD card
insert, connect KB and USB (not powered on), power up pi3
expanded filesystem (raspi-config), set regions (US), KB, etc
changed default pw

**THEN**
turned up USB drive

pi@pi3:~$ sudo blkid
/dev/sda1: UUID="4A22-8003" TYPE="exfat" PARTUUID="4d0cae1c-01"

pi@pi3:~$ sudo fdisk -l

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    92159    83968   41M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      92160 62333951 62241792 29.7G 83 Linux

Disk /dev/sda: 698.7 GiB, 750156373504 bytes, 1465149167 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x4d0cae1c

Device     Boot Start        End    Sectors   Size Id Type
/dev/sda1          63 1465144064 1465144002 698.7G  7 HPFS/NTFS/exFAT

**changed to ext4**

pi@pi3:~$ sudo mkfs.ext4 /dev/sda1 -L untitled
mke2fs 1.42.12 (29-Aug-2014)
/dev/sda1 contains a exfat file system
Proceed anyway? (y,n) y
Creating filesystem with 183143000 4k blocks and 45793280 inodes
Filesystem UUID: 1d9f1645-803a-4cd5-9066-5ed3f83486ca
Superblock backups stored on blocks: 
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
        4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968, 
        102400000

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done     

pi@pi3:~$ sudo blkid
/dev/sda1: LABEL="untitled" UUID="1d9f1645-803a-4cd5-9066-5ed3f83486ca" TYPE="ext4" PARTUUID="4d0cae1c-01"

**NEXT**
pi@pi3:~$ sudo nextcloudpi-config
Downloading updates                                                                                                                                                                                                                        
Performing updates
NextCloudPi updated to version v0.12.6
NextCloudPi configuration ver.  v0.12.6

**enabled fail2ban
get this err exiting from that section:
/usr/local/bin/nextcloudpi-config: line 73: unexpected EOF while looking for matching `''                                                                                                                                                  
/usr/local/bin/nextcloudpi-config: line 95: syntax error: unexpected end of file**

**enable nc-automount**

**moved ncdatadir**
**DATADIR       /media/USBdrive/ncdata**
**moved ncdatabase**
**DBDIR         /media/USBdrive/ncdatabase**

set nc-limits to 2GB file size
[... 195 lines cut ...]
total 48
drwxrwxr-x+ 5 root     root      4096 May 26 22:35 .
drwxr-xr-x  3 root     root      4096 May 26 02:20 ..
drwxrwxr-x+ 2 root     root     16384 May 26 01:51 lost+found
drwxrwxr-x+ 8 www-data www-data  4096 May 26 02:40 ncdata
drwxrwxr-x+ 5 mysql    mysql     4096 May 26 03:33 ncdatabase
lrwxrwxrwx  1 root     root        15 May 26 22:35 USBdrive -> /media/USBdrive

compare to
pi@pi3:~$ ls /media/USBdrive -all
total 32
drwxr-xr-x 5 root     root      4096 May 26 02:22 .
drwxr-xr-x 3 root     root      4096 May 26 02:20 ..
drwx------ 2 root     root     16384 May 26 01:51 lost+found
drwxr-x--- 8 www-data www-data  4096 May 26 02:40 ncdata
drwxr-xr-x 5 mysql    mysql     4096 May 26 03:00 ncdatabase

**rebooted and now web GUI is back**

**NOW, I went into web GUI as admin, deleted the 2 users I created, then recreated them.**

**NOW I get GUI, and can login as both users AND creat folders and upload files as either user on web GUI and phone app.
FIXED (not sure how I did this plus permissions still dont match original (drwxrwxr-x+ 8 VS drwxr-x--- 8 for www-data ncdata directory)  but its working.**  (I guess I could get this back to match original but I'm not linux proficient so will have to go research that :) - do you see any risks with the extra permissions security wise?
Still not sure where I went wrong but hopefully this may help another novice get things working if they run into my issues, it seems likely you need to move the ncdata and ncdatabase dirs to usb drive BEFORE creating any ancillary users (or perhaps just create, delete, and recreate users is enough?
Thanks!

nachoparker, 2017-05-28:
Hi,

Thank you very much. All that detailed testing is really useful for me!!!

Thanks to this, I detected two little bugs, fixed in v0.12.12 and v0.12.13. One of them was caused by you having `pi` own the database and datadir but the tool `nextcloudpi-config` should be robust to that ;)

Your final permissions are ok.

Note that the fact that the user `pi` does not have permissons does not imply that there is a permissions problem, as long as `www-data` and `mysql` can access their folders.

Good job!

closed 2017-05-28 by nachoparker

karpaz, 2017-10-05:
>burn 12 version to SD card  
>insert, connect KB and USB (not powered on), power up pi3
What is KB  @clanger7

karpaz, 2017-10-05:
Can anybody tell me how to do it easier ?
-  I installed clear nextcloudpi 12.0.3 on Raspberry pi 3 type 
- Connected LAN, keyboard, microSD, Hdmi
- connect to 192.168.1.X from my PC2 (other PC in same network)
- on PC2 write to browser that ip with 192.168.1.X:4443
- write user, and pass and in that update nextcloudpi
- Now my nextcloud is allright, BUT when i log to nextcloud from PC2 i have cca 400Mb remaining capacity storage for admin user, (my card in raspi is 16Gb) that's problem number one.

When i enable automount, connect HDD named (**disk**) formated to Ext4 to USB and in WEB GUI in "EXTERNAL MEMORY" write mount location "/media/**disk** the button is now green, but when i open it in files i don't have permission. Please make some tutorial for this. if it is maked please tell me the link to that. I cannot find it on forum or GitHub.

BUGS/ or MISTAKES
1 - In user-web (192.168.1.X) in setting app "External storage support" still show green button also if it connected wrong. !

2 - When you don't enable automount in webgui terminal(192.168.1.X:4443) or raspi terminal and try move nc-datadir or database, it doesn't inform you that you don't mounted. It move nc-datadir or database to "black hole" or i don't know where...... ,but on usb isn't anythink.....