nextcloud/nextcloudpi #15: Latest Nextcloudpi crashes due to PHP error

sunjam, 2017-06-03:
Flashed latest NextcloudPi 12.0 image onto a pi2, enabled ssh, plugged into my router via ethernet and disabled https.
NextCloudPi updated to version v0.12.15
Enabled nc-automount for external usb disk
Moved data and database to the usb disk
Updated nc-limits to 20gb.
Successfully deployed Nextcloud via the browser and started installing some apps.
Added Linux desktop client to a laptop and began syncing a few files and folders up to 5gb in size for about 12gb data total.
Nextcloud server crashed during the sync and wouldn't re-connect via the browser or ssh.
`status-php server replied: Internal Server Error` popped up on the desktop Nextcloud client under settings.
I hard rebooted the Pi and am able connect over ssh, but not via the browser or desktop client.
Typing in the pi's ip only shows a white screen in Firefox or HTTP 500 error in Chromium.
[Here is the output from Nextcloud.log](https://gist.github.com/sunjam/81f9a50ee16dbc57041eb07a96cf7b4d)

nachoparker, 2017-06-06:
Thanks for the feedback.

Are apache2, mysql and php-fpm processes up and running? check the status of mysqld ?

What is the output of
`pgrep mysqld`
`pgrep apache2`
`pgrep php-fpm7.0`

sunjam, 2017-06-07:
pi@raspberrypi:~ $ pgrep mysqld
pi@raspberrypi:~ $ pgrep apache2
691
31131
31132
pi@raspberrypi:~ $ pgrep php-fpm7.0
494
692
693
pi@raspberrypi:~ $ sudo /etc/init.d/mysql start
[ ok ] Starting mysql (via systemctl): mysql.service.
pi@raspberrypi:~ $ pgrep mysqld
2294
2432

My instance immediately kicked back up, asking me to log back in on my laptop's desktop client and resumed syncing files.

nachoparker, 2017-06-07:
It seems that the fact that your database was moved to  the USB drive made it crash which leads me to think that there was a problem reading or writing from your drive.

Did you try rebooting the raspberry pi?

If this ever happens again, it would be nice to capture the database logs to corroborate
`tail /var/log/mysql/error.log`

If this is the case, I am afraid that there is not much I can do from my side. If the USB drive is not stable enough it might be better to leave the database in the SD card.

I will close now.  Please, don't hesitate to add more information, or reopen if my guess is wrong.

closed 2017-06-07 by nachoparker

joeynewman, 2017-06-12:
It's my first issue and hopefully the comment find place at the right position.
I have the same problem like sunjam that a white page with http500 error is shown.
But before that nc runs a week without problems. 1st part of the problem was that when I visit the nc-page in the browser that the loginproces doesn't work. It only brings me back to the loginpage. after beeing sure that it is no CAPS-Lock / typing error I connect via ssh.
It says that a update is availaible and i performed sudo ncp-update
After that the white page with http500 error comes up.
I looked for extdrive mount. Everything as it should be.
In my newbiw panic mode I tried 
sudo apt-get -y update
sudo shutdown - r 0
without any success.

looking in the forum found this post so here are my answers to the point you asked for:

pi@klaut:~ $ pgrep mysqld
pi@klaut:~ $ pgrep apache2
715
730
731
pi@klaut:~ $ pgrep php-fpm7.0
615
716
717
pi@klaut:~ $ sudo /etc/init.d/mysql start
[....] Starting mysql (via systemctl): mysql.serviceJob for mysql.service failed. See 'systemctl status mysql.service' and 'journalctl -xn' for details.
 failed!

I hope you can help me?! Is there someting elso u need to identifiy the problem?
Is it possible to change the nc-data dir back to the sd card? How is the path for that?

regards 
joey

nachoparker, 2017-06-12:
First, can you paste the output of `systemctl status mysql.service`?

Then, try rebooting your raspberry pi with `sudo reboot` and repeat the `pgrep` procedure. Is it now up?

If you ever want to move your database back to the SD card, run `sudo nextclodpi-config`, then navigate to `nc-database` and move it back to the original path `/var/lib/mysql`

Also, do you have fail2ban enabled? If you tried too many times with the wrong password you might get temporarily locked out. Read the relevant post [here](https://ownyourbits.com/2017/02/24/nextcloudpi-fail2ban-installer/)

Don't panic, by breaking things is how we learn, plus there is many people online that will help you ;)

joeynewman, 2017-06-12:
pi@klaut:~ $ systemctl status mysql.service
â mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql)
   Active: failed (Result: exit-code) since Mon 2017-06-12 04:10:11 UTC; 12h ago
  Process: 987 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)

after rebooting

I think, this looks better.

pi@klaut:~ $ pgrep mysqld
770
920
pi@klaut:~ $ pgrep apache2
934
961
962
pi@klaut:~ $ pgrep php-fpm7.0
688
935
936

what does the numbers say?

after that, it still don't work. But I found the mistake. 

FYI, I use fail2ban but that wasn't the reason for the first issue.
the problem was the nc-directory. I was wondering u wrote nc-database because i only moved nc-datadir to ext drive. So I looked what's there and here was one old path that doesn't exist.

I used a cheap USBdrive for the first attempt and beacause of automount it become USBdrive and I changed it after that to a external SSDDrive that is USBdrive3. But because of not knowing the original path i helped myself with a little workaround that I umount and mount the SSDdrive manuel to Media3 after every reboot.  

But with my motivation to fix that I type faster than I could think. -.-
I tried to put the directory to another usbdrive that I'm able to relocate the SSD to Media that my workaround isn't longer necessary. But not knowing what exactly I do with changing the directories I'm back to my whitescreen and also with 

Launching nc-database
Job for mysql.service failed. See 'systemctl status mysql.service' and 'journalctl -xn' for details.
Done. Press any key...

pi@klaut:~ $ systemctl status mysql.service
â mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql)
   Active: failed (Result: exit-code) since Mon 2017-06-12 17:28:22 UTC; 2min 38s ago
  Process: 1921 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)

pi@klaut:~ $ journalctl -xn
No journal files were found.

buhuuu. 

I could imagine that a additional sidepunch is also that maybe the powersupply for 2 usbdrive at the same time maybe is to much and shutdown one or both?

How can I change my dumpass moves to a smart solution? Maybe there is a much easier way than I try?
And after that, I have to learn how to backup the nc and be prepared for restore ;)

Thanks a lot for your help and your work with this incredible Nextcloud Project!!

joey

nachoparker, 2017-06-13:
Yes, it seems that you messing with the hard drives while NextCloudPi was working did not make it happy.

Check out the documentation about automount here

https://ownyourbits.com/2017/05/24/nextcloudpi-gets-automount-backups-nextcloud-online-installation-nextcloud-12-and-more/

You just need to use a label in your hard drive so you can reference it in `nextcloudpi-config` as `/media/hddlabel` instead of `/media/USBdriveX`

joeynewman, 2017-06-13:
after hours at the office thinking about other stuff I retried it and did it one by one. 
Tadaa! Everything runs!
Thanks for you help and the good documentation and generally about your time you spend to your Project! Absolutly incredible!!

regards
joey

nachoparker, 2017-06-13:
awesome

d-popov, 2018-01-03:
I just had the same problem.
Fresh installation on Rpi_1, changed all the passwords.
v0.44.2 
This is the output of the tail command: 
`
{"reqId":"Wk0tZX8AAQEAAAKcOj0AAAkA","level":3,"time":"2018-01-03T19:22:14+00:00","remoteAddr":"192.168.1.100","user":"admin","app":"PHP","method":"GET","url":"\/index.php\/settings\/apps","message":"Class 'OC\\App\\AppStore\\Fetcher\\Fetcher' not found at \/var\/www\/nextcloud\/lib\/private\/App\/AppStore\/Fetcher\/AppFetcher.php#31","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; AS; rv:11.0) like Gecko","version":"12.0.4.3"}
{"reqId":"Wk0wS38AAQEAAAKcOnIAAAwT","level":3,"time":"2018-01-03T19:34:37+00:00","remoteAddr":"192.168.1.100","user":"admin","app":"PHP","method":"GET","url":"\/ocs\/v2.php\/apps\/activity\/api\/v2\/activity\/all?format=json&previews=true&since=0","message":"TypeError: Argument 5 passed to \u00b4\u00a9%::__construct() must be an instance of {, instance of OCA\\Activity\\ViewInfoCache given at \/var\/www\/nextcloud\/apps\/activity\/lib\/Parameter\/Factory.php#71","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; AS; rv:11.0) like Gecko","version":"12.0.4.3"}
{"reqId":"Wk0wvn8AAQEAAAKcOnUAABEX","level":3,"time":"2018-01-03T19:37:56+00:00","remoteAddr":"192.168.1.100","user":"admin","app":"PHP","method":"GET","url":"\/index.php\/apps\/gallery\/","message":"Class 'Leafo\\ScssPhp\\Formatter' not found at \/var\/www\/nextcloud\/3rdparty\/leafo\/scssphp\/src\/Formatter\/Crunched.php#22","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; AS; rv:11.0) like Gecko","version":"12.0.4.3"}
`
It says something about a missing "Fetcher"