nextcloud/nextcloudpi #16: Backup

fzrkid, 2017-06-04:
This is not a good way to start my first question ever on GitHub/nextcloudpi but I am trying to get my nextcloud install to perform a backup. 
I installed the latest image running NC12 from ownyourbits. I finally managed to the system working with my data directory and database moved to the external USB drive. 

I tried following the many tutorials for backup  but it never seems to work. 

My USB storage drive path for NC is  /media/ncstore/ncdata, I have both the data files and database files there. What I am not able to do is copy that to another external device.

I am relatively new to linux and NC, but can usually fumble my way eventually. Not sure if did this correctly, but I tried naming the drive ncstore so I could identify the drives by name but that doesnt seem to be working. I did the partitioning on my liux machine with Gparted. Partitioned all the drives to Ext4.

I tried using Rsync to copy the directories from my NC USB drive to another USB drive but it doesnt work. I dont recall the error message ( I will post when I get  a change to try again). 

Will the new nextcloud pi image have the ability to run backup with the new USB data directory? Would be nice if it could copy the directory name when creating the new storage location and make that default.

I am super excited about running my nextcloud pi / raspberry pi server. I have wanted to do something like this for years.

Trying to get all my methods working first then I will build a fresh install.

Would it be possible to perform the backup to a network shared drive, example if I plug another external USB drive into my router?

I appreciate the help. Looking forward to a full implementation of NextcloudPi so I can cut the cord to onedrive/gdrive

nachoparker, 2017-06-05:
Hello,

Did you use the `nc-automount` feature or did you set up your USB mount in `/etc/fstab`?

> I tried naming the drive ncstore so I could identify the drives by name but that doesnt seem to be working

Are you meaning to say that you used the same label for two different drives? Did you try to plug them in at the same time?

You could use `nc-backup` and `nc-restore` for your purposes. See [this](https://ownyourbits.com/2017/05/24/nextcloudpi-gets-automount-backups-nextcloud-online-installation-nextcloud-12-and-more/)

If you want to do it more manually, rsync should work, as long as you keep the original permissions for the datadir and database folders.

You could definitely backup to a network mounted storage, or FTP the backup, or anything like that. Only that you have to set that up yourself manually.

At this moment, your datadir will *not* be included in the backup if you moved it with `nc-datadir`, only your configuration, apps, etc...

If you have ideas of features that would be great for you and are missing, please share them!

Finally, this is not really a specific technical question or bug report about nextcloudpi, so I would like to suggest that we continue this conversation in the official forums, maybe some other person read it and  have nice ideas or suggestions to make :)

https://help.nextcloud.com/c/support/appliances-docker-snappy-vm

fzrkid, 2017-06-05:
Thank you for the reply. 

I used the nc-automount to mount the USB drive for storage.

I used ncstore for my storage of the NC data and DB while I named my backup drive ncbackup. 

Not to continue to clutter this thread I will copy the question and reply over at the NC discussion from the link you provided.

I look forward to being part of the NC community, I hope I can some worthwhile input.

Regards,
fzrkid

closed 2017-06-05 by fzrkid