nextcloud/nextcloudpi #17: nc-datadir didn't work for me

CarlosElViper, 2017-06-05:
Hello,
I used the NextCloudPi image to install and all worked great, thanks!  Then I ran nc-automount to auto mount my WD Pidrive, all is good, but  when I ran nc-datadir I got this error:
Launching nc-datadir
./nc-datadir.sh: line 34: [: too many arguments
data directory Your data directory is invalid
Please check that the data directory contains a file ".ocdata" in its root.

Cannot create "data" directory
This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:145
Stack trace:
#0 /var/www/nextcloud/console.php(91): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main} not found

I ran also ran: 
pi@raspberrypi:/media/USBdrive $ ls -la
total 48
drwxrwxr-t 9 root  pi     4096 Jun  2 14:02 .
drwxr-xr-x 3 root  root   4096 May 30 18:12 ..
drwxr-xr-x 2 root  root   4096 May 31 18:30 chuck
drwxr-xr-x 2 root  root   4096 Dec 31  1969 data
drwxr-xr-x 2 root  root   4096 Dec 31  1969 images
drwx------ 2 root  root  16384 Dec 31  1969 lost+found
drwxr-xr-x 5 mysql mysql  4096 Jun  1 17:35 ncdatabase
drwxr-xr-x 4 root  root   4096 Dec 31  1969 shared
drwxr-xr-x 2 root  root   4096 Dec 17 20:18 tmp
lrwxrwxrwx 1 root  root     15 May 28 15:30 USBdrive -> /media/USBdrive
pi@raspberrypi:/media/USBdrive $ 

So I think I have something somewhere gone bad.
Also  I am running this through SSH Pi@xxx.xxx.x.xx on  my local network.

I am kind new at this type of thing so your patience is appreciated.
thanks
carlos

nachoparker, 2017-06-05:
Hello,

Thanks for your feedback.

What is the nextcloudpi version you are using? I see a couple things happening that were fixed in v0.12.13 and v0.12.14.

If you could start over from copying the SD card, and run `ncp-update` to get to v0.12.15 we can start troubleshooting this.

Again, thanks for the feedback and patience!

nachoparker, 2017-06-05:
I will close it now because I suspect it was fixed in said version updates.

Don't hesitate to continue the conversation here, and please reopen if the problem persists

closed 2017-06-05 by nachoparker

CarlosElViper, 2017-06-07:
Hello again,
I have done as you instructed and I am to the stage where I will be moving the data location to my USBdrive.  I think that I would like to leave the database on the SC card (for Speed) and the data on my USBDrive.  Does this sound like a reasonable approach?
thanks

nachoparker, 2017-06-07:
Hello,

It sounds good to me! It depends on how careful you want to be with wearing down the SD, and the speed of both SD and USB drive.

For sure, it is more 'safe' in the SD card if for instance you forget to plug the USB drive in or something, NC will still work.

CarlosElViper, 2017-06-08:
Hello again,
I did not get any errors, so I think I'm good.
thanks again

nachoparker, 2017-09-05:
Duplicate of #14