nextcloud/nextcloudpi #18: Doesn't work well on iPhone app

jeffkz, 2017-06-05:
Work ok for awhile after fresh restart of server. But after several concurrent transferring it just crashed.
After I turned on the "Automatic upload of camera images" it crashed the server with the below settings:
------------------------------------
- Upload camera photos/videos: ON
------------------------------------
- Upload camera photos: ON
- Only use Wi-Fi connection: ON
------------------------------------
Upload camera videos automatically
------------------------------------
Upload camera videos: ON
Only use Wi-fi connection: ON
------------------------------------
Upload images in the background: Off
------------------------------------
UPLOAD ALL CAMERA PHOTOS/VIDEOS: OFF
(THIS ONE CAUSE MAJOR HANG TO THE APP BECAUSE I HAVE MORE THAN 1000+ Images and Videos in the gallery)
------------------------------------
Use Subfolder: ON

nachoparker, 2017-06-06:
What do you mean crash the server? Could you provide some logs? What are the steps to reproduce?

Thanks for feedback

nachoparker, 2017-07-09:
It has been a month. I will close this now.

closed 2017-07-09 by nachoparker