nextcloud/nextcloudpi #21: Can not connect to NextCloudPi from local LAN

RalfJahns, 2017-06-05:
I installed NextCloudPi but when I open the IP of the the Pi in the browser (from PC in local LAN), it gives me the following message (translated from german by me and my english is not that good):

[image]

You access the server from a not trusted domain.

Please contact your administrator. If you are the Administrator of this instance, configure the “trusted_domains”-setting in config/config.php. A sample configuration …
If your configuration allows, as administrator you can use the following button, to set this domain as a trusted domain.

I don’t understand that, so I clicked the button.

Then my firefox complains that it can not connect!?
The browser shows the following URL:
https://localhost/index.php/settings/admin?trustDomain=nextcloudpi

Don't understand why it shows "localhost"?!?

nachoparker, 2017-06-05:
Hello,
Thanks for the feedback.

It seems like you are using DynDNS. Is that right?
Did you use your DynDNS domain in `letsencrypt` in `nextcloudpi-config`?
Did you *NOT* activate `dnsmasq` from `nextcloudpi-config`?

If all of the above is correct, I think I know what is happening. Please confirm.

nachoparker, 2017-06-06:
I have included a fix for this use case. Please, update to v0.12.16 with `sudo ncp-update` and retry.

I'll now close this. Feel free to reopen if the problem persists

closed 2017-06-06 by nachoparker

RalfJahns, 2017-06-06:
Huge thank you for your quick action. Unfortunately I still have exactly the same problem :-(

You asked if I used my DynDNS domain in letsencrypt and that is exactly what I did. You also asked if I activated dnsmasq and I didn't do it.

One more thing: When I access my raspberry from the local LAN, my browser shows this URL: https://nextcloudpi/index.php because nextcloudpi is the network device name of my raspberry. And this is exactly the domain, that is shown in the screenshot above (the one I could add as a trusted domain by pressing the button). When I try to access the raspberry from outside (mobile phone without WLAN) I use the URL https://**********.dyndns.org and then the button (in the screenshort) shows exactly this URL as the untrusted domain that I could add.

Google took me to a few sites, that told me I should add something to the trusted_domains section in the config.php but I don't want to break your fine distribution by doing modifications I don't really understand.

nachoparker, 2017-06-06:
I recommend that you use *****.dyndns.org from inside your house as well. It is the most convenient setup, for instance to tell your Android apps how to access your cloud independently of wether you are at home or outside.

You can do this from `sudo nextcloudpi-config` dnsmasq section.

Read also this https://ownyourbits.com/2017/03/09/dnsmasq-as-dns-cache-server-for-nextcloudpi-and-raspbian/