nextcloud/nextcloudpi #22: Installation problems with letsencrypt, certbot, nextcloud and IPv6

TheNachtschatten, 2017-06-06:
Sorry for bad English, I'm German)
Please Note: I replaced the IPv6 addresses with aaaa:aaaa:... or bbbb:bbbb:... in the following text, so that the text is easier to read.
Hello,
I'm not sure if this is the right place to ask (or the letsencrypt forum), but I have trouble setting up letsencrypt using the nextcloudpi image.

What I did:
1. I downloaded the nextcloudpi image from here: https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/ and copied it to an sd card
2. I configured my router in a way that Port 80 won't be blocked (IPv6 port forwarding).
3. I ran "$ sudo ip addr add aaaa:aaaa:aaaa:aaaa:cccc:cccc:cccc:cccc/64 dev eth0" on my server.
4. I used ddnss.de to acquire a domain
5. I ran "sudo nextcloudpi-config" -> "letsencrypt" -> "yes" -> "< mydomain >.ddnss.de" ->"start". 
Here is the output of the command:
Launching letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Obtaining a new certificate
Performing the following challenges:
tls-sni-01 challenge for < mydomain >.ddnss.de
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. < mydomain >.ddnss.de (tls-sni-01): urn:acme:error:malformed :: The request message was malformed :: no working IP addresses found for "< mydomain >.ddnss.de"

IMPORTANT NOTES:
- The following errors were reported by the server:

Domain: < mydomain >.ddnss.de
Type: malformed
Detail: no working IP addresses found for "< mydomain >.ddnss.de"

To fix these errors, please make sure that you did not provide any
invalid information to the client, and try running Certbot again.
- Your account credentials have been saved in your Certbot
configuration directory at /etc/letsencrypt. You should make a
secure backup of this folder now. This configuration directory will
also contain certificates and private keys obtained by Certbot so
making regular backups of this folder is ideal.
Done. Press any key...

If I enter my domain in the browser bar, I get a notification that something is wrong with the certificate. 
I would be thankful if anyone knows why this problem occures or how to fix it.

Additional information that may be useful:

OS: Raspbian
Root acces to server?: Yes

Please note: I have not configured IPv4 port forwarding and the IPv6 configuration is a bit messed up: 
Running ifconfig on the server at first bootup to see what's the IPv6 address produced the following output:
inet6 addr: aaaa:aaaa:aaaa:aaaa:bbbb:bbbb:bbbb:bbbb/64 Scope:Global
inet6 addr: fe80::cccc:cccc:cccc:cccc/64 Scope: link

I tried configure the router to forward aaaa:aaaa:aaaa:aaaa:bbbb:bbbb:bbbb:bbbb but that didn't worked so I configured the router to forward aaaa:aaaa:aaaa:aaaa:cccc:cccc:cccc:cccc and ran "sudo ip addr add aaaa:aaaa:aaaa:aaaa:cccc:cccc:cccc:cccc/64 dev eth0" on the server. Now the output of ifconfig is:
inet6 addr: aaaa:aaaa:aaaa:aaaa:cccc:cccc:cccc:cccc/64 Scope:Global
inet6 addr: aaaa:aaaa:aaaa:aaaa:bbbb:bbbb:bbbb:bbbb/64 Scope:Global
inet6 addr: fe80::cccc:cccc:cccc:cccc/64 Scope: link

This site: http://www.ipv6now.com.au/pingme.php is able to ping both aaaa:aaaa:aaaa:aaaa:cccc:cccc:cccc:cccc and < mydomain >.ddnss.de

nachoparker, 2017-06-06:
Thanks for the feedback. I will link to this conversation in the let's encrypt forums.

https://community.letsencrypt.org/t/problems-with-letsencrypt-certbot-nextcloud-and-ipv6/35484

Let me know if you need any information from the nextcloudpi side

nachoparker, 2017-06-07:
Quoting from the letsencrypt forums:

> UPDATE: IT WORKS!
> The problem was that I enabled IPv6 portforwading at port 80 , but not 443. I enabled both port 80 and port 443 and now everything works fine.

closed 2017-06-07 by nachoparker