nextcloud/nextcloudpi #23: Can not connect to NextCloudPi

RalfJahns, 2017-06-06:
Huge thank you for your quick action. Unfortunately I still have exactly the same problem :-(
Unfortunately I can't reopen the issue, so I have to open a new one with the same title. Sorry for that. As far as I read "you cannot re-open your own issues if a repo collaborator closed them".

You asked if I used my DynDNS domain in letsencrypt and that is exactly what I did. You also asked if I activated dnsmasq and I didn't do it.

One more thing: When I access my raspberry from the local LAN, my browser shows this URL: https://nextcloudpi/index.php because nextcloudpi is the network device name of my raspberry. And this is exactly the domain, that is shown in the screenshot above (the one I could add as a trusted domain by pressing the button). When I try to access the raspberry from outside (mobile phone without WLAN) I use the URL https://**********.dyndns.org and then the button (in the screenshort) shows exactly this URL as the untrusted domain that I could add.

Google took me to a few sites, that told me I should add something to the trusted_domains section in the config.php but I don't want to break your fine distribution by doing modifications I don't really understand.

Hope for more help from you.

Best regards
Ralf

nachoparker, 2017-06-06:
I answered to you here, please let's keep the conversation in that thread

https://github.com/nextcloud/nextcloudpi/issues/21

closed 2017-06-06 by nachoparker

nachoparker, 2017-09-05:
Duplicate of #21