nextcloud/nextcloudpi #24: Q: Using my own subdomain

RyanTG, 2017-06-07:
Thanks for this great tool! It definitely smooths out the process of using/installing NC.

I have a really basic question about domains. So instead of using no-ip, I just want to use my own subdomain. So traditionally with NC and apache, to use my own subdomain, I would edit /etc/apache2/sites-available/nextcloud.conf (right?). But with NextCloudPi, am I supposed to instead use dnsmasq? Or do I do both?

Sorry for my newb question. I always get stuck on DNS stuff...

nachoparker, 2017-06-07:
Hello,
You can do either. The extras in `nextcloudpi-config` are provided for convenience but are not mandatory

If you want to use your own domain, you have to register it in `/var/www/nextcloud/config/config.php`. That step is automatically performed by `nextcloudpi-config`

closed 2017-06-07 by nachoparker

RyanTG, 2017-06-07:
K. Thanks. I just didn't know if dnsmasq was the same thing as registering it manually, or if it did more, or what. 

I don't recall seeing that step in `nextcloudpi-config`, but perhaps it was present when I first ran it and I opted to do it later. And then I couldn't find where to add it.

Sidenote: I was about to make a new "possible bug" issue for this: When I use nc_datadir, it works great, BUT afterward there isn't a checkmark next to the name in the list. Super minor, and perhaps intentional.

nachoparker, 2017-06-07:
You can learn more here

https://ownyourbits.com/2017/03/09/dnsmasq-as-dns-cache-server-for-nextcloudpi-and-raspbian/

Sidenote: it is not a "service" that remains active, so I figured that the checkmark didn't apply 8P

Thanks for the feedback

RyanTG, 2017-06-07:
That makes sense. But I was paranoid, and was like, "is the external storage definitely working?" And had to navigate to the data folder to double-check that the files were in there.

Thanks for your help!

nachoparker, 2017-06-07:
haha thx, I'll think about it

RyanTG, 2017-06-19:
Man, I still can't get my own domain to work. My router goes through a VPN, which may be affecting things.

1. Pointed the A Record of my subdomain (cloud.mysweetsite.com) to my DHCP ip address.
2. Forwarded a port on my VPN, using local port 443.
3. In my router (dd-wrt), I set up this iptable:

````
iptables -I FORWARD -i tun1 -p udp -d 192.168.1.102 --dport 15119 -j ACCEPT
iptables -I FORWARD -i tun1 -p tcp -d 192.168.1.102 --dport 15119 -j ACCEPT
iptables -t nat -I PREROUTING -i tun1 -p tcp --dport 15119 -j DNAT --to-destination 192.168.1.102:443
iptables -t nat -I PREROUTING -i tun1 -p udp --dport 15119 -j DNAT --to-destination 192.168.1.102:443
````

15119 is the port I forwarded on my vpn.

4. in /etc/apache2/sites-available/nextcloud.conf I'm specifying ServerName cloud.mysweetsite.com

Whaaaaat am I doing wrong!

nachoparker, 2017-06-19:
Sorry, that seems out of scope for this issue tracker. 

I suggest you try in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)