nextcloud/nextcloudpi #25: Web site unavailable after reboot

jsfrederick, 2017-06-09:
Installed NextcloudPI  and all was working fine.  The next day, I rebooted the RPi, and I an unable to connect to the web page.  I just get a blank with screen in Safari, Firefox, and Chrome.

I am not a very experienced in web servers and services, so I do not know where to start.  Please let me know what additional info you need for troubleshooting.

Thanks!

jsfrederick, 2017-06-12:
Never mind...

I saw a previous issue (#15) that talks about the same issue.  Looks like my external hard disk powers down.  I'll need to reset it to be always on, or get a new one.

Thanks!

closed 2017-06-12 by jsfrederick

nachoparker, 2017-06-12:
Please, let us know about your USB drive model, and if you find a way of keeping it awake. ( in #15  )

In any case, if you identified your problem, you might want to keep your database in the SD card.

jsfrederick, 2017-06-12:
The current drive its a WD Elements external usb drive.  Not sure I can keep it powered on, but and trying to figure that out.  The new drive will be a WD Red, that one is on all the time so I should not have an issue.  

Any thoughts on how bit the "database" my grow to? I have my boot drive as a 32GB USB stick.  I'm using this RPi as NOTHING but a NextcloudPi server so I have most of that 32GB available for the database.  My data is on the 4TB external drive.

Thanks again! NextcloudPi is AWESOME!

nachoparker, 2017-06-13:
The database does not contain files, just information about them, so you do not need a big SD card to hold it. 4GB is enough.

You are welcome!

jsfrederick, 2017-06-16:
Ok, thanks.  Got my new WD red hard drive and installed.  Everything is working fine.  Will keep an eye on it for a few days.  The WD red drive is a NAS drive that is always on, no power down.

jsfrederick, 2017-06-19:
Happened again today.  I decided to move the database BACK to the USB Stick I boot from.  We will see if the fixes it.

nachoparker, 2017-06-20:
:(
What power supply are you using?

jsfrederick, 2017-06-22:
Using the 3A power supply from Western Digital.  Moved the database back to the USB stick that I boor from and all is well.  I do think that the database DOES need to be on the boot drive, for an unknown reason.

Thanks again for NextcloudPi.  It's the BOMB!

nachoparker, 2017-06-22:
Haha, thank you.

Well, it has worked for me for many years with 3 different hard drives. I suspect many HDDs end up going to standby mode at some point.

I will include a warning about this in `nc-data`

Thanks for the feedback.

jsfrederick, 2017-06-24:
I'm pretty sure that the drive does not go into standby, I will have to do some tests...

Now that I've move the database back the the boot drive, everything is working fine.  DOn't know if we can ever really know the root cause.  The message in nc-database is probably a good idea.

Thanks again for all your work on this!
NextCloudPi ROCKS!

nachoparker, 2017-09-05:
Duplicate of #15