nextcloud/nextcloudpi #26: can nextcloud pi be updated through the ui?

JunkyardCat, 2017-06-11:
or is it still broken?

nachoparker, 2017-06-12:
It can, only that in some cases it has been found to fail. This depends on many things, like your configuration or installed apps.

In order to minimize risks, it is recommended to backup your instance first with `nc-backup`.

See [here](https://ownyourbits.com/2017/05/24/nextcloudpi-gets-automount-backups-nextcloud-online-installation-nextcloud-12-and-more/) for more info on `nc-backup` and `nc-restore`

closed 2017-06-12 by nachoparker

JunkyardCat, 2017-06-13:
whats a docker?

nachoparker, 2017-06-13:
A docker container is a technology to *pack* software such as NextCloudPi. You can google it.

There will still be images just the same, it is just an alternative method for getting NextCloudPi. I will support both from now on.

Please, open a different topic for each question so other people can benefit from the answers ;)