nextcloud/nextcloudpi #29: USB HHD not showing up

themtbeerbottle, 2017-06-16:
Got the image loaded and followed directions to a T.   Formatted hard drive to ext4.   I'm using a docking station for the hard drive.  It just doesn't seem to show up.  Lsusb shows nothing.  Any advice would be greatly appreciated. Thanks Mike

nachoparker, 2017-06-17:
does it show in `lsblk`?

themtbeerbottle, 2017-06-17:
First of all thank you for the response and help.  I truly appreciate it. 

so after running the command lsblk i was able to see the 1TB usb hard drive.  I figured I must have obviously done something wrong.  I read through the instructions again and realized the error of my ways (I think)

before I run the "nc-automount" I need to have the hard drive removed like it says in the prompt on the screen.  

Then 

Before I run the "nc-datadir" command I need to have the usb hard drive attached and showing up with the lsblk command.  (I am not very smart .  If there is a way to mess this stuff up I find it)

I was NOT mounting the hard drive before running the nc-datadir

Looks like it is good to go.  
Thanks again for your help and a great setup of nextcloud on a PI.  

Cheers
Mike

nachoparker, 2017-06-17:
> before I run the "nc-automount" I need to have the hard drive removed like it says in the prompt on the screen.

Should not be necessary to have it removed at the time of `nc-automount`.

I am glad that it works

closed 2017-06-17 by nachoparker