nextcloud/nextcloudpi #35: Login Image not appearing after install

Shadowstreik, 2017-06-30:
Hello. I have installed NextCloudPi based on the OwnYourBits instruction. Everything runs great. However, the login image does not appear. Instead, only a solid background color appears. This applies to the default login image and any other image file I select to serve as that image.

I have run the LetsEncrypt process, but am not certain if that would have any effect.

Would there be a permissions issue at play, here?

Shadowstreik, 2017-07-02:
Navigate to...

> /var/www/nextcloud/config

enter the config file...

> sudo nano config.php

Find the trusted domains...

```
  'trusted_domains' => 
  array (
    0 => 'localhost',
    1 => '192.168.xxx.xxx',
    4 => 'letsencrypt domain',
```

Add the domain of the actual IP the subdomain points to...

>     2 => 'actual.domain.letsencrypt.points.to',

Save file and exit from Nano.

Do not simply restart the Apache webservice, but a warm reboot will suffice.

Issue is resolved. Closing issue.

nachoparker, 2017-07-03:
Hi,
I am glad you got it working.

I do not understand very well... isn't it `2 => actual.domain` the same as ` 4 => letsencrypt.domain`? or am I missing some configuration that you have?

Also, I am interested if anybody else run into this.

Thanks for your feedback

nachoparker, 2017-07-03:
I'll reopen this until we clarify. Maybe there is something else that can be done from NCP to avoid this situation.

Shadowstreik, 2017-07-04:
>  isn't it 2 => actual.domain the same as 4 => letsencrypt.domain?

I would think it is, or should be, just by how I understand it. Though, I would enter the letsencrypt.domain and be met with the solid background color, only. It wasn't until I did an entry with actual.domain that the logo and login image would appear. I would agree that the letsencrypt.domain should be sufficient.

I also notice during my syncs that the same issue would be encountered for larger file syncs. But, I think that just may be a performance limitation by the RPi3, itself.

Shadowstreik, 2017-07-04:
Update- I just went to the letsencyrpt.domain and the login background image and logo image are not there. It is only the background color. I own the domain I'm using for the letsencrypt.domain, not that that would make any difference/"should".

Shadowstreik, 2017-07-04:
[image]

Shadowstreik, 2017-07-04:
Update- I shut off Fail2Ban and the images load, now. I'm going to watch it the next couple of days. If it still shows, Fail2Ban may be the problem.

nachoparker, 2017-07-06:
fail2ban... doesn't sound likely.

Do you have `modsecurity` on?

nachoparker, 2017-07-06:
I was re-reading...

I still can't understand the *subdomain* thing. Is that what you input in `nextcloudpi-config` in the `letsencrypt` section?

Shadowstreik, 2017-07-08:
In the LetsEncrypt selection of the NCP Config utility, I entered the address for NCP which includes the sub.

Shadowstreik, 2017-07-08:
Performed a full rebuild Activated Fail2Ban. Result:
[image]

Disabled Fail2Ban. Result:
[image]

Conclusion: Fail2Ban was/is the issue.

nachoparker, 2017-07-09:
good one

does it happen only from outside? or does it also happen using your local network IP?

Could you have a look at the logs for fail2ban, apache and nextcloud?

I want to reproduce it myself

nachoparker, 2017-07-09:
finally, I could reproduce it 

thanks

nachoparker, 2017-07-09:
It seems like I have found the root cause. CSP code does not play well with PHP cache.

I have reported the issue to the server developers ( https://github.com/nextcloud/server/issues/5664 )

The most likely outcome is that I will have to blacklist some files from PHP cache.

Does not seem `fail2ban` related. According to my testing, it can be worked around for now by issuing these commands

```
systemctl stop php7.0-fpm
systemctl stop mysqld
systemctl start php7.0-fpm
systemctl start mysqld
```
or rebooting. The problem is that then, if we try to access through IP again (after we last did through domain) we would be in the same situation.

It is a minor issue though, as the user will most likely access the same way always

nachoparker, 2017-07-14:
This has been fixed upstream. Will be included in NC 12.0.1

Will close now, thanks for reporting and please, reopen if it still happens after update.

closed 2017-07-14 by nachoparker