nextcloud/nextcloudpi #37: Swap file move questions

jsfrederick, 2017-07-05:
Not so much an issue as some questions.  I used the NCPi-config to move my swap file to my external hard disk.  That works fine.  Now, when I use the "free -t" command, it lists swap 0 Total, 0 used, and 0 free.  I think that Raspbian is NOT recognizing the move of the swap file.  I do see that the dphys-swapfle config file for swap does point the the correct location and is the correct size.

Just wondering if you are seeing the same thing. I am asking because I installed RPi-Monitor and it displays the swap file info.  It get's that data from /proc/meminfo.  on my machine that  shows 0 for both total and available swap.  I would like to get RPi-Montior working with the correct  swap info if possible.

nachoparker, 2017-07-05:
what's the output of 

`swapon -s`

and 

`sudo systemctl status dphys-swapfile`

Does the file exist in disk? what are the permissions?

A quick test showed me that `swapon` and `test -f` report the correct thing, even after a reboot.

jsfrederick, 2017-07-05:
[swapcommand.txt](https://github.com/nextcloud/nextcloudpi/files/1126056/swapcommand.txt)

[image]

jsfrederick, 2017-07-05:
If I do a dphys-swapfile swapon after reboot, it works.  But swap is turned off on reboot.

nachoparker, 2017-07-06:
`sudo systemctl status dphys-swapfile` ?

jsfrederick, 2017-07-06:
root@nextcloudpi:~# sudo systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Wed 2017-07-05 17:33:33 EDT; 20h ago
  Process: 490 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 05 17:33:33 nextcloudpi systemd[1]: Starting LSB: Autogenerate and use a swap file...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: Starting dphys-swapfile swapfile setup ...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: want /media/USBdrive3/var/swap=1024MByte, generating swapfile .../sbin/dphys-swapfile: ERROR: swap file /media/USBdrive3/var/swap missing! you need to first run  /sbin/dphys-swapfile setup  to generate one
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: done.
Jul 05 17:33:33 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.

jsfrederick, 2017-07-06:
root@nextcloudpi:~# swapon -s
Filename                                Type            Size    Used    Priority
/media/USBdrive3/var/swap               file            1048572 0       -1
root@nextcloudpi:~#

jsfrederick, 2017-07-06:
root@nextcloudpi:~# cat /proc/meminfo
MemTotal:         994232 kB
MemFree:          332264 kB
MemAvailable:     757920 kB
Buffers:           60232 kB
Cached:           435168 kB
SwapCached:            0 kB
Active:           448184 kB
Inactive:         165652 kB
Active(anon):     135052 kB
Inactive(anon):    19264 kB
Active(file):     313132 kB
Inactive(file):   146388 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:       1048572 kB
SwapFree:        1048572 kB
Dirty:               124 kB
Writeback:             0 kB
AnonPages:        118428 kB
Mapped:            78140 kB
Shmem:             35888 kB
Slab:              34436 kB
SReclaimable:      22384 kB
SUnreclaim:        12052 kB
KernelStack:        2224 kB
PageTables:         2452 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     1545688 kB
Committed_AS:    2025328 kB
VmallocTotal:    1064960 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
CmaTotal:           8192 kB
CmaFree:            6800 kB
root@nextcloudpi:~#

jsfrederick, 2017-07-06:
So I am really confused.  Swap file is there:

root@nextcloudpi:~# ls -l /media/USBdrive3/var
total 1048580
-rw------- 1 root root 1073741824 Jun 16 10:30 swap
root@nextcloudpi:~#

nachoparker, 2017-07-06:
Can you try

`sudo systemctl status dphys-swapfile`

and then

`systemctl status dphys-swapfile `

?

jsfrederick, 2017-07-06:
root@nextcloudpi:~# sudo systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Wed 2017-07-05 17:33:33 EDT; 20h ago
  Process: 490 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 05 17:33:33 nextcloudpi systemd[1]: Starting LSB: Autogenerate and use a swap file...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: Starting dphys-swapfile swapfile setup ...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: want /media/USBdrive3/var/swap=1024MByte, generating swapfile .../sbin/dphys-swapfile: ERROR: swap file /media/USBdrive3/var/swap missing! you need to first run  /sbin/dphys-swapfile setup  to generate one
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: done.
Jul 05 17:33:33 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.

root@nextcloudpi:~# systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Wed 2017-07-05 17:33:33 EDT; 20h ago
  Process: 490 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 05 17:33:33 nextcloudpi systemd[1]: Starting LSB: Autogenerate and use a swap file...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: Starting dphys-swapfile swapfile setup ...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: want /media/USBdrive3/var/swap=1024MByte, generating swapfile .../sbin/dphys-swapfile: ERROR: swap file /media/USBdrive3/var/swap missing! you need to first run  /sbin/dphys-swapfile setup  to generate one
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: done.
Jul 05 17:33:33 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
root@nextcloudpi:~#

nachoparker, 2017-07-06:
You typed the same command twice.

Do the **start** one, and then the **status** one

Also, surround your output by three backticks (without the spaces), before and after the code

```
` ` `
```

at the beginning and the end to format your code ;)

jsfrederick, 2017-07-06:
Ok, got it now...
```
root@nextcloudpi:~# systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Wed 2017-07-05 17:33:33 EDT; 21h ago
  Process: 490 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 05 17:33:33 nextcloudpi systemd[1]: Starting LSB: Autogenerate and use a swap file...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: Starting dphys-swapfile swapfile setup ...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: want /media/USBdrive3/var/swap=1024MByte, generating swapfile .../sbin/dphys-swapfile: ERROR: swap file /media/USBdrive3/var/swap missing! you need to first run  /sbin/dphys-swapfile setup  to generate one
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: done.
Jul 05 17:33:33 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
root@nextcloudpi:~#
```

jsfrederick, 2017-07-06:
```
root@nextcloudpi:~# swapon -s
Filename                                Type            Size    Used    Priority
/media/USBdrive3/var/swap               file            1048572 0       -1
root@nextcloudpi:~#
```

jsfrederick, 2017-07-06:
```
root@nextcloudpi:~# sudo systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Wed 2017-07-05 17:33:33 EDT; 21h ago
  Process: 490 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 05 17:33:33 nextcloudpi systemd[1]: Starting LSB: Autogenerate and use a swap file...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: Starting dphys-swapfile swapfile setup ...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: want /media/USBdrive3/var/swap=1024MByte, generating swapfile .../sbin/dphys-swapfile: ERROR: swap file /media/USBdrive3/var/swap missing! you need to first run  /sbin/dphys-swapfile setup  to generate one
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: done.
Jul 05 17:33:33 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
root@nextcloudpi:~#```

```
root@nextcloudpi:~# systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Wed 2017-07-05 17:33:33 EDT; 21h ago
  Process: 490 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 05 17:33:33 nextcloudpi systemd[1]: Starting LSB: Autogenerate and use a swap file...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: Starting dphys-swapfile swapfile setup ...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: want /media/USBdrive3/var/swap=1024MByte, generating swapfile .../sbin/dphys-swapfile: ERROR: swap file /media/USBdrive3/var/swap missing! you need to first run  /sbin/dphys-swapfile setup  to generate one
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: done.
Jul 05 17:33:33 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
root@nextcloudpi:~#
```

jsfrederick, 2017-07-06:
I did not see a start command in any of your messages.  Just the status commands.  Did I miss something?  

```
root@nextcloudpi:~# swapon
NAME                      TYPE  SIZE USED PRIO
/media/USBdrive3/var/swap file 1024M   0B   -1
root@nextcloudpi:~#
```

nachoparker, 2017-07-06:
lol, I'll copy the relevant message again

> Can you try
> 
> sudo systemctl start dphys-swapfile
> 
> and then
> 
> systemctl status dphys-swapfile
> 
> ?

jsfrederick, 2017-07-06:
Thank you!  I really do appreciate you assistance!

```
root@nextcloudpi:~# sudo systemctl start dphys-swapfile
root@nextcloudpi:~# systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Wed 2017-07-05 17:33:33 EDT; 23h ago
  Process: 490 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 05 17:33:33 nextcloudpi systemd[1]: Starting LSB: Autogenerate and use a swap file...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: Starting dphys-swapfile swapfile setup ...
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: want /media/USBdrive3/var/swap=1024MByte, generating swapfile .../sbin/dphys-swapfile: ERROR: swap file /media/USBdrive3/var/swap missing! you need to first run  /sbin/dphys-swapfile setup  to generate one
Jul 05 17:33:33 nextcloudpi dphys-swapfile[490]: done.
Jul 05 17:33:33 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
Jul 06 14:53:10 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
Jul 06 14:53:26 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
Jul 06 17:31:10 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
Jul 06 17:31:39 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
root@nextcloudpi:~#
```

nachoparker, 2017-07-06:
Sorry, it should have been a restart

`sudo systemctl start dphys-swapfile`

and then

`systemctl status dphys-swapfile`

jsfrederick, 2017-07-06:
When I execute /sbin/dphys-swapfile setup, it says this:

```
root@nextcloudpi:~# /sbin/dphys-swapfile setup
want /media/USBdrive3/var/swap=1024MByte, checking existing: keeping it
```

jsfrederick, 2017-07-06:
```
root@nextcloudpi:~# sudo systemctl restart dphys-swapfile
root@nextcloudpi:~# systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Thu 2017-07-06 17:39:06 EDT; 8s ago
  Process: 9778 ExecStop=/etc/init.d/dphys-swapfile stop (code=exited, status=0/SUCCESS)
  Process: 9790 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 06 17:38:58 nextcloudpi dphys-swapfile[9790]: Starting dphys-swapfile swapfile setup ...
Jul 06 17:38:58 nextcloudpi dphys-swapfile[9790]: want /media/USBdrive3/var/swap=1024MByte, checking existing: keeping it
Jul 06 17:39:06 nextcloudpi dphys-swapfile[9790]: done.
Jul 06 17:39:06 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
```

nachoparker, 2017-07-06:
ok, so if you start it while it's running, it detects and uses the swapfile.

During boot though, it seems like it doesn't. Maybe the USB drive is not ready yet?

How do you mount it ? automount? fstab?

jsfrederick, 2017-07-06:
Using your default mount process from Nextcloud Pi.  Here is my FSTAB, swap is on USBdrive 3.
I guess I could move swap back to the USB drive that I boot from??

```
proc            /proc           proc    defaults          0       0
PARTUUID=24eafa51-01  /boot           vfat    defaults          0       2
PARTUUID=24eafa51-02  /               ext4    defaults,noatime  0       1

# Rules for automounting both at boot and upon USB plugin. Rely on udev rules
/dev/USBdrive  /media/USBdrive         auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive1 /media/USBdrive1        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive2 /media/USBdrive2        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive3 /media/USBdrive3        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive4 /media/USBdrive4        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive5 /media/USBdrive5        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive6 /media/USBdrive6        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive7 /media/USBdrive7        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive8 /media/USBdrive8        auto    defaults,noatime,auto,nofail    0       2
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```

nachoparker, 2017-07-06:
so, you have `nc-automount` active, right?

jsfrederick, 2017-07-06:
Yes.  Trying to move swap back to /var right now.  looks like that may have crashed the Pi...

jsfrederick, 2017-07-06:
Ok, moved it back to /var.  Looks like this now.  I'm going to let it run this what for a while and see what happens.

Thanks again.

```
root@nextcloudpi:/var# swapon -s
Filename                                Type            Size    Used    Priority
/var/swap                               file            1048572 0       -1
root@nextcloudpi:/var# systemctl status dphys-swapfile
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Thu 2017-07-06 17:51:43 EDT; 4min 8s ago
  Process: 2902 ExecStop=/etc/init.d/dphys-swapfile stop (code=exited, status=0/SUCCESS)
  Process: 2913 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 06 17:49:10 nextcloudpi dphys-swapfile[2913]: Starting dphys-swapfile swapfile setup ...
Jul 06 17:51:43 nextcloudpi dphys-swapfile[2913]: want /var/swap=1024MByte, generating swapfile ... of 1024MBytes
Jul 06 17:51:43 nextcloudpi dphys-swapfile[2913]: done.
Jul 06 17:51:43 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
root@nextcloudpi:/var#
```

nachoparker, 2017-07-06:
Well, that would never be able to crash the pi by itself.

If your pi crashed it is surely a hardware thing, either power, SD card or something.

In any case. If we want to test wether your USB drive is not ready at the time of mounting and that is why it fails, we could alter fstab and try a couple things out.

If you are up for it, tell me (along with the output of `lsblk -f`).

Good night

nachoparker, 2017-07-09:
I'll close this. Let me know if you want to test this further, and we can continue here

jsfrederick, 2017-07-09:
That will work, we can do some testing whenever you have the time.  It's not an urgent matter, but I would love to understand what the issue is.

Thanks much!  LOVE me some NextcloudPi!

nachoparker, 2017-07-09:
great!

so what's the output of `blkid`?

jsfrederick, 2017-07-09:
```
root@nextcloudpi:~# blkid
/dev/sda1: LABEL="boot" UUID="70CE-EB76" TYPE="vfat" PARTUUID="24eafa51-01"
/dev/sda2: UUID="f2100b2f-ed84-4647-b5ae-089280112716" TYPE="ext4" PARTUUID="24eafa51-02"
/dev/sdb1: LABEL="clouddisk" UUID="0207aff9-c275-4e72-adc2-3aee1070f79e" TYPE="ext4" PARTUUID="cd5d65a5-b002-42d4-8497-3826bdf574a7"
root@nextcloudpi:~#
```

nachoparker, 2017-07-09:
then let's try mounting it without automount and see if that affects things.

Disable `nc-automount`

Then, edit `fstab` with 

```
sudo nano /etc/fstab
```

add this line

```
UUID=0207aff9-c275-4e72-adc2-3aee1070f79e      /media/USBdrive3 ext4 defaults,noatime,norelatime,async     00
```

Then, reboot and try moving the swap over there again.

Reboot and check stuff again... `swapon -s`, `/proc` ...

jsfrederick, 2017-07-09:
Ok, done  here are the outputs:
```
root@nextcloudpi:~# swapon -s
Filename                                Type            Size    Used    Priority
/media/USBdrive3/var/swap               file            1048572 0       -1
root@nextcloudpi:~#

root@nextcloudpi:~# systemctl status dphys-swapfile -l
● dphys-swapfile.service - LSB: Autogenerate and use a swap file
   Loaded: loaded (/etc/init.d/dphys-swapfile)
   Active: active (exited) since Sun 2017-07-09 19:18:44 EDT; 1min 36s ago
  Process: 1946 ExecStop=/etc/init.d/dphys-swapfile stop (code=exited, status=0/SUCCESS)
  Process: 1958 ExecStart=/etc/init.d/dphys-swapfile start (code=exited, status=0/SUCCESS)

Jul 09 19:18:44 nextcloudpi dphys-swapfile[1958]: Starting dphys-swapfile swapfile setup ...
Jul 09 19:18:44 nextcloudpi dphys-swapfile[1958]: want /media/USBdrive3/var/swap=1024MByte, checking existing: keeping it
Jul 09 19:18:44 nextcloudpi dphys-swapfile[1958]: done.
Jul 09 19:18:44 nextcloudpi systemd[1]: Started LSB: Autogenerate and use a swap file.
root@nextcloudpi:~#
```

nachoparker, 2017-07-10:
thanks!!

Well, that looks good. It seems like after reboot it was able to access the swapfile on USBdrive3. It seems like automount might not make the USB mount ready by the time the swap service tries to access it.

Correct me if I am wrong
 - With automount only it fails after reboot
 - With an entry on fstab it doesn't

I have been testing myself a little bit and I have found that moving the swapfile does not play well with automount.

I'll reopen into a bug

Many thanks

jsfrederick, 2017-07-10:
Yes, at this point your two assumptions are correct.  I will keep watching this and update you if anything changes.

Thanks!

JimmyKater, 2017-07-28:
hey...
i seem to have a mount-problem as well using nc-automount. (no swap-file on USB, only ncdata)
so i'm trying it by editing /etc/fstab - the way you suggested above.
seems to be working.
though i can't disable automount. the check stays there

JimmyKater, 2017-07-28:
ha! found out how to disable automount from nc-config.

nachoparker, 2017-07-28:
@JimmyKater 

haha nice

So... what was your problem with nc-automount?

I put the automount feature together pretty fast and it needs polishing. I figured it would help the very basic case for unexperienced people, but more work needs to be done to support several partitions and stuff like that.

At the moment, if you want a more  advanced setup you have to deal with manually editing `/etc/fstab`, the classic way. Maybe we should just use udisks.

JimmyKater, 2017-07-31:
i dunno what my problem with nc-automount was. but it seems that i'm better off without using it. just wanted to let you know that there was a problem... ;)
usually i'm with you on the point to support unexperienced users.

nachoparker, 2017-07-31:
fixed in `v0.18.4`

closed 2017-07-31 by nachoparker

doodhout, 2017-09-17:
Just FYI, I suffered from the same problem as OP on NextCloudPi v0.28.1.

My setup: Raspberry Pi 1 rev B+, 8GB µSD-card and 32GB high performance USB flash drive where I had moved data, database and swap.

Symptoms and analysis:
After reboot I noticed nextcloud wasn't working any more.
I checked and found that /etc/init.d/mysql had failed to start.
Then I checked mount and I saw that the USB flash drive was mounted under the normal path, but with a 1 as a suffix. The regular path existed as well, but with only the swap file on it.
That led me to believe the creation of the swap file was causing issues with the USB flash drive being properly mounted and thus causing the database service to fail because it could not address its database through the expacted path.

I moved swap back to /var/swap, implemented the fstab fix, changed swap file path in config back to USB flash drive and without restarting the swap service I rebooted.

This solved my problem, but I wonder why it kept happening even though you marked this fixed in an earlier release?

Anyway, thanks for this project, setting up NextCloud has become so much nicer compared to a few years ago.

nachoparker, 2017-09-18:
Hi,

Thanks for the feedback

I am pretty sure I tested moving the swap, but I'll test again. 

What is the fstab fix you mention?

doodhout, 2017-09-18:
The solution - fix is a too strong word - was to use fstab instead of nc-automount, just like you recommended to @jsfrederick a few months ago.

nachoparker, 2017-09-18:
I see, hopefully when all the issues are sorted out that won't be necessary ;)