nextcloud/nextcloudpi #38: Let's Encrypt certificate auto renewall, How ?

TheTiPi, 2017-07-05:
I’ve enable Let’s Encrypt, resulting a certificate valid for only two month.
How do I renew (automaticly) the certificate ?
Do I have to “sudo nextcloudpi-config” every two month ?

Shadowstreik, 2017-07-06:
It sounds like you are looking for a method in which the process of auto-renewal is fully automated and not involving any manual effort, at all. Please, correct me if I am wrong.

This article from ZDNET may be worth a peek. It is very telling...  http://www.zdnet.com/article/lets-encrypt-brings-wildcard-certificates-to-the-web/

nachoparker, 2017-07-06:
If you used `nextcloudpi-config` it already configures NextCloudPi to renew the certificate every month automatically. No need to do anything.

@Shadowstreik thx for the info ;)

closed 2017-07-06 by nachoparker

TheTiPi, 2017-07-06:
That's what I did.
Thanks