nextcloud/nextcloudpi #40: Adding 2nd user seems to create instability

Matthze, 2017-07-06:
Hi,

I encountered massive Problems, when adding a second user to my Nextcloud. 
Prior to adding a second user the system seemed to work fine.
After adding a second user, the log in for the second user did never work reliable.
Error Messages started to come up sporadically for the first user "admin" as well.

Following more details:
Hope they are sufficient to reproduce. Let me know if you need logs, including path where to find them.

---------------

First time this behaviour was encountered with a system which started at v.0.12 and which was continously upgraded to v.0.16.0
Second time with a system upgraded to v.0.16.1
Third time with a system upgraded to v.0.16.0

Below the exact sequence of events with the third try.

---------------

Flashed image NextCloudPi_06-28-17.tar.bz2 to SD card

Booted Raspberry Pi (Ethernet, keyboard and HDMI Monitor connected)

called sudo raspi-config
- changed localization settings (timezone, keyboard and UTF code)
- changed password
- enabled SSH
- expand filesystem 
- reboot
-------------------

logged into nextcloud (admin, ownyourbits)
admin user: Selected settings - admin => worked 
admin user: Selected settings - apps => worked
admin user: Added a new user at settings - users (name "test", password, nothing else)
admin user: Selected settings - admin => worked 
admin user: Selected settings - apps => worked
Logged in on a different computer for new user "test" => received error message 2., tried second time, received error message 3.
admin user: Selected settings - Personal => received error message 1.
admin user: deleted user "test"
admin user: Selected settings - Personal => received error message 1.
admin user: created user "test"
Logged out with user admin
Logged in with user test => received error message 2., tried second time, received error message 3.
Rebooted
logged into nextcloud (admin, ownyourbits)
admin user: Selected settings - Personal => worked
Logged in on a different computer for new user "test" => received error message 2., tried second time, received error message 3.

-------------------

- Used Opera on one Win Computer and Firefox and Chromium on Linux computer.
- cleared Browsercache => Still same observations

- First time problem was encountered error messages were received with settings - admin, second time with settings - admin and settings - apps. I.e. the menue options which were not used prior to creating a new user.

-------------------
Browser Error Message 1 (appeared with tries 1 and 2)

This page isn’t working

192.168.178.33 is currently unable to handle this request.
HTTP ERROR 500

-------------------	
Browser Error Message 2

Service Unavailable

The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.

Apache/2.4.25 (Raspbian) Server at 192.168.178.33 Port 443

-------------------	
Browser Error Message 3

Internal Server Error
The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.

-------------------	

Output of pgrep sqld => 888
Output of pgrep apache2 => 902, 921, 922
Output of pgrep php-fpm7.0 => 738, 1383, 1385, 1386

nachoparker, 2017-07-06:
what's the output of 

`sudo tail -f /var/log/mysql/error.log /var/log/apache2/*.log /var/www/nextcloud/data/nextcloud.log`

nachoparker, 2017-07-12:
nevermind, I am unable to reproduce it anymore, and I have tried several times

Aside from doing it manually, I have tried to automate the steps. See this testing code

https://github.com/nextcloud/nextcloudpi/blob/master/tests/tests.py

If anybody finds this issue again, please reopen. Additionally, it would be awesome if said person could extend that example to be able to reproduce the issue. 

It is I think easy to read and extend in python

closed 2017-07-12 by nachoparker