nextcloud/nextcloudpi #43: Cant get letsencrypt working

deljones, 2017-07-07:
Hi all..

Finally finally got port forwarding to work for my NCpi installation which is now working very well... Spending to much time on it though exploring!!Love it!

I have two remaining issues....
1. LetsEncrypt. I have used the Auto Encryption in raspi-config. If I put my public address in the domain box I don't get any errors but when I got to my site the site is still not secure? If I put the domain that I have registered with noip, cloudwalker.webhop.me, I get errors telling me to check I have a public IP address etc.

Which ever I choose I either get errors no errors but an unsecured site. Driving me crazyyyyy!

2. The famous cloaking of the URL at NOIP....
If I don't cloak my URL then I can get to my site using my url cloudwalker.webhop.me externally but it shows the full public ip address and port number. If I cloak the URL in NOIP I get a blank page returned.

I'm clearly a dope and clearly missing something...

D

nachoparker, 2017-07-07:
Hi,
Good job.

You are supposed to enter the domain that you got from noip in the `letsencrypt` section of `nextcloudpi-config`.

Could you try the process again and paste here those errors?

One question... did you open the ports yourself, or through `nextcloudpi-config`?

I have not tried the URL cloacking... will look at it.

deljones, 2017-07-07:
Hi there...  Thanks for your quick reply...

So I start port forwarding in raspi-config. To start with my 4th trusted domain in my NC config is my public address and port number. I'm using 8888 because 443 is used by my NAS. the 8888 is redirected to 443.

[image]

So that's all good... I switched off upnp as instructed on my modem. Created the service, NCBox on 8888 then on the firewall opened 8888 pointing to the NCBox. I can reach the server outside using public the address and inside using the LAN ip. I created a hostname at noip pointing to my public address which works except I have to have the 'Mask URL' unchecked otherwise I get a blank page. The masking of the URL is supposed to hide your public address and only show the hostname blah blah blah....

Regarding letsencrypt..
Started the utility in raspi-config

[image]

Entered the hostname, email address of my account. After a while I get this...

[image]

When I press 'Any key' the no-ip hostname has overwritten trusted domain 4 in my NC config, which it is supposed to do which was my public address and port number with the new hostname

[image]

I think I've done everything right?

D

nachoparker, 2017-07-09:
Hi, I'm back

Well, you are trying to use let's encrypt with a non-standard port. Only port 443 for HTTPS is works for Let's Encrypt for security reasons. Here is the [info](https://community.letsencrypt.org/t/support-for-ports-other-than-80-and-443/) and the [vulnerability](https://www.ietf.org/mail-archive/web/acme/current/msg00524.html)

If you think about it, Let's encrypt is expecting to access your NextCloudPi on 443 but in your case 443 is the NAS. That is why it times out

See the explanation [here](https://ownyourbits.com/2017/03/17/lets-encrypt-installer-for-apache/)

It seems like you will have to use the DNS challenge for that, which is more involved.

I'll close it but we can continue talking ;)

closed 2017-07-09 by nachoparker

deljones, 2017-07-09:
Hi

Yes I am a dope!! I realised what I had done but I also did something 
else completely dumb... my noip hostname was set as a 'URL', ie redirect 
rather than an 'A' record. I ran the letsencrypt script again once I had 
corrected that small error and it worked...

I now have a fully working secured instance at 
digitalcarnage.webhop.me.....

Sadly I had to reinstall the instance 3 times today. If I move either of 
the swapfile or Dbase to my external drive I get a blank error 500 page 
on reboot :(

The current build has everything on a fast 16G SD card except the ncdata 
which is on an external USB drive...

Thanks ever so much for you help and understanding. I can tell you that 
in the past 36 hours I have leant so much thanks to you and this 
project... Its been a time vampire :)

Keep it up

Del

<author@noreply.github.com>
Sent: 09/07/2017 19:05:33

nachoparker, 2017-07-09:
The USB drive needs to be on and responsive at all times... no sleeping, no stand by. See #15 
Haha, imagine how much time this is consuming for me 😃