nextcloud/nextcloudpi #44: USB backup

fzrkid, 2017-07-07:
Would be great to see an auto back-up option to back up the USB data drive to another external USB drive. Something that would act as an easy recovery if the original usb data drive fails.
This would include the option to remove the backup drive and keep it for safe storage and periodically plug in for the backup.

In addition to data and db backup, it would be nice to have a rolling backup for the RPI image. Something like a mirror image of Nextcloud PI, data and db that makes for an easy recovery.

Apologies if this is not posted in the correct forum. Still trying to get the hang of github

nachoparker, 2017-07-07:
thanks, I have been requested this before.

I will enhance the `nc-backup` feature when I have time ;)

deljones, 2017-07-12:
Not very elegant I know, but would it be safe to shutdown, remove the SD and and HDD and then clone them to new drives that you can stash away 'just in case'?

You'd probably have to do this every week or so.. I have my NC setup just how I want it but dreading the SD or HD failing over...

DJ

nachoparker, 2017-07-12:
totally fine! You can do that if you want. A bit manual and repetitive but totally safe.

Auto backups will help. I also have some ideas for what you want. 

In my case, I automatically mirror the contents of the USB drive once a week. I will share that "solution" as well sooner or later in the blog.

;)

deljones, 2017-07-12:
Excellent 

Del

t: +44 7957140512

e: derek@derekjones.net

w: derekjones.net

fzrkid, 2017-07-13:
Great, looking forward to the update. Been looking for a solution like this for many years.. I think part of my newbie self challenge is getting the hang of permissions for the HDD mount points and how to define the mount points if the drives are not attached to the RPI in the same order. Did that make sense?

Would it be helpful if I provided a short write-up of how I envision this happening within  nextcloudpi configuration tool?

I can envision this being like a simple turnkey package where anyone could make like a RPI_nextcloud box. simply burn the image, plug in the drives, assign a name to the drives like drive_1 and drive_2 (if they want a backup) and be off and running in 15 minutes.

How about something that can write to the config file, so when/if someone wants to add their domain they can do it in the config tool instead of having to remember where the config file and and dealing with the not so friendly text editing tools.

I am a win user that has converted to linux, so I am accustomed to the win tools. Not saying they are better, just maybe a little more intuitive. :)

I will stop. Excited to see the update.

Regards,
fzrkid

nachoparker, 2017-07-13:
awesome, thanks for the ideas.

They are very useful for me.

fzrkid, 2017-07-14:
I have many ideas and tweaks that I think can be helpful. Lets see this iteration and I will do my best to screw it up then go from there.

fzrkid, 2017-07-26:
Using the new web interface and trying to understand how linux/nextcloud manages these things.

The data directory is now /media/USBdrive/ncdata

I partitioned the drive with the web interface and named it USBDrive_1.  Why doesnt the config.php show "/media/usbdrive/usbdrive_1/ncdata" or something to that effect to help with managing the USB drives.

TO help instill some confidence(at least for me) the paths/directories are correct it would be nice to have some sort of text/visual cue indicating where the data will get stored. Even having a Disk capacity tool, just to keep watch on how much free space there is left would be nice.

Sorry for these newbie sort of questions. I have looked online and played around, but it still confuses me. I am too accustomed to windows partitions

nachoparker, 2017-07-27:
hi John!

Could you copy and paste this into a new issue that I'll label as a 'question'. This way we can keep that conversation separated from this 'feature request'?

Thanks! I'll explain over there ;)

nachoparker, 2017-07-31:
This has been implemented in `v0.18.0`

The feature is called `nc-backup-auto`, and you can schedule a backup every X days.

You can if you want specify the destination dir to a different USB drive. If you use labels and `nc-automount`, you can select the destination USB drive by label, for example in `nextcloudpi-config`

```
DESTDIR /media/mysecondHDDlabel/
```

Hope it helps! Let me know what you think ;)

closed 2017-07-31 by nachoparker

nachoparker, 2017-08-10:
> What happens to the configuration if I remove the 2nd drive that is used for the backup, will nc-backup-auto retain the settings?

The backup will fail, because `nc-backup-auto` will try that location that is not there any more.
It will try it blindly and if the drive is not there it will just fail.

Like I said before, right now the automount feature needs some love. It does create unnecessary folders and symlinks at this moment :s

fzrkid, 2017-08-10:
Ok, I will keep plugging away at it and send anything I find.

Super excited about this project and enjoy watching it evolve.

nachoparker, 2017-08-10:
awesome, thank you!