nextcloud/nextcloudpi #45: Be able to execute nc-backup via commandline

brianekummer, 2017-07-07:
It'd be useful to be able to run nc-backup by commandline, so that it could be scheduled using cron, such as for scheduling weekly full backups.

nachoparker, 2017-07-09:
Added to the wishlist!

I am still undecided about wether provide a command or just enhance the `nc-backup` feature to allow scheduled backups.

nachoparker, 2017-07-31:
It is now possible to do this from version `v0.18.0`.

Ideally, you would use the new `nc-backup-auto` feature, where you can specify a periodic backup.

If you want to use it on cron or scripts you can

 - 1 . Configure the backup from `nc-backup-auto`. You can keep it inactive, but still the configuration parameters will be saved (like backup destination dir).
 - 2.1. Once configured, trigger it manually from `sudo /usr/local/bin/ncp-backup`, or
 - 2.2. Once configured, trigger it from systemctl with `sudo systemctl start nc-backup`

closed 2017-07-31 by nachoparker