nextcloud/nextcloudpi #46: Change Default TCP Port (apache2)

sruckh, 2017-07-07:
How can the default TCP (https:443) port be changed?

I modified the file /etc/apache2/sites-available/nextcloud.conf and changed the line --
<VirtualHost _default_:443>
to
<VirtualHost _default_:4343>

enabled the new site and reloaded apache2 configuration.

This did not seem to be sufficient.  I am not sure if there are any other rules (ie., like fail2ban) that need to be updated, or any other configuration files that need modification, but making only the above configuration change did not seem to be sufficient.

Thank You.

sruckh, 2017-07-07:
Adding another 'Listen' line to /etc/apache2/ports.conf with the above port helped.  Looks like some of the images are missing from the web page, but login works from both web page (PC) and also from NextCloud app for android.

nachoparker, 2017-07-09:
cool, thanks for letting us know.

I guess I could make HTTPS port configurable, but it would not work with let's encrypt ( #43  )

Probably better to leave it as it is for now, but I am open to suggestions

closed 2017-07-09 by nachoparker