nextcloud/nextcloudpi #47: Failure to load after updates

Shadowstreik, 2017-07-07:
Today...
- ran sudo apt-get update & upgrade on RPi's Raspi side
- updated NCP to version 0.16.2
- went to ip of ncp, received a solid white screen (Firefox). Chrome reports 500 error.

The service is currently unusable.

Shadowstreik, 2017-07-08:
Performed only expedient and efficient solution, a full rebuild from scratch.

Shadowstreik, 2017-07-09:
Two more rebuilds after running into the same problem. And now, there's more. It gave a little text...

> Internal Server Error
> 
> The server encountered an internal error and was unable to complete your request.
> Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
> More details can be found in the server log.

No technical details are given. It doesn't say where that log is.

More rebuild(s) on the way, tomorrow.

nachoparker, 2017-07-09:
Maybe this thread will help

https://github.com/nextcloud/nextcloudpi/issues/15

Thanks again for your testing

Shadowstreik, 2017-07-09:
Hi. Naw, I didn't. I left the database alone. The only think not actually part of the Pi, itself, is the data storage on a USB drive.

I did a build, this morning, am running V0.16.2, but haven't run apt-get update nor upgrade and it's been working flawlessly as I sync data back onto the unit.

At this point, I'm skiddish about running apt-get update and upgrade and just relying on ncp-update for relevant updates.

nachoparker, 2017-07-09:
ok, the images come with automatic security updates, so you should be fine.

I'll test this soon

nachoparker, 2017-07-09:
`v0.16.2`

Just tested this and I could access no problem.

If anybody runs into this again, please reopen and post steps to reproduce

closed 2017-07-09 by nachoparker