nextcloud/nextcloudpi #48: External Storage App broken?

GitHub-23666, 2017-07-08:
With ssh and mount it WORKS, can access all my stuff. No Problems!
- sudo mkdir /media/PRIVAT
- sudo mount -t cifs //xxx.xxx.xxx.xxx/PRIVAT /media/PRIVAT -o user=xyz,pass=xyz

Under External Storage App only red square?
- FOLDER         PRIVAT
- HOST		xxx.xxx.xxx.xxx
- SHARE		PRIVAT
- USER		xyz
- PASSWORD	xyz

My Harware:
- RPI 2
- 64GB SDCARD
- Image: NextCloudPi_06-28-17.img

nachoparker, 2017-07-10:
```
# smbclient
Segmentation fault
```
O_O

Even after purge && apt-get update && apt-get install smbcilent I am getting this

Seems like Raspbian `smbclient` is broken

nachoparker, 2017-07-12:
I found the culprit, this is an update on the progress of investigation 

`smbclient` works perfectly well until you install `mariadb` in [this line](https://github.com/nextcloud/nextcloudpi/blob/master/lamp.sh#L52)

It's so weird. `strace` hints at some memory violation accessing at `/dev/random`

<details>
<summary>strace output</summary>

```

pi@raspberrypi:~ $ strace smbclient
execve("/usr/bin/smbclient", ["smbclient"], [/* 16 vars */]) = 0
brk(0)                                  = 0x5666e000
uname({sys="Linux", node="raspberrypi", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76f54000
access("/etc/ld.so.preload", R_OK)      = 0
open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=42, ...}) = 0
mmap2(NULL, 42, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0x76f53000
close(3)                                = 0
open("/usr/lib/arm-linux-gnueabihf/libarmmem.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0h\5\0\0004\0\0\0"..., 512) = 512
lseek(3, 17960, SEEK_SET)               = 17960
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
lseek(3, 17696, SEEK_SET)               = 17696
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\3\f\1\22\4\24"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0644, st_size=18920, ...}) = 0
mmap2(NULL, 83236, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76f12000
mprotect(0x76f17000, 61440, PROT_NONE)  = 0
mmap2(0x76f26000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0x76f26000
mprotect(0x7e9b0000, 4096, PROT_READ|PROT_WRITE|PROT_EXEC|PROT_GROWSDOWN) = 0
close(3)                                = 0
munmap(0x76f53000, 42)                  = 0
open("/usr/lib/arm-linux-gnueabihf/samba/tls/v7l/neon/vfp/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/tls/v7l/neon/vfp", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/tls/v7l/neon/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/tls/v7l/neon", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/tls/v7l/vfp/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/tls/v7l/vfp", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/tls/v7l/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/tls/v7l", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/tls/neon/vfp/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/tls/neon/vfp", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/tls/neon/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/tls/neon", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/tls/vfp/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/tls/vfp", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/tls/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/tls", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/v7l/neon/vfp/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/v7l/neon/vfp", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/v7l/neon/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/v7l/neon", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/v7l/vfp/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/v7l/vfp", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/v7l/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/v7l", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/neon/vfp/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/neon/vfp", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/neon/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/neon", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/samba/vfp/libpthread.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/samba/vfp", 0x7e9af0c0) = -1 ENOENT (No such file or directory)
[... cut ...]
fcntl64(3, F_GETFD)                     = 0
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFCHR|0666, st_rdev=makedev(1, 9), ...}) = 0
getrusage(0x1 /* RUSAGE_??? */, {ru_utime={0, 50000}, ru_stime={0, 100000}, ...}) = 0
getpid()                                = 13840
read(3, "\22\216]\366m\262D2\5\366:\216\220P\216\252[\263\360\177\227_)\355\220T\"\224\367\266tF", 32) = 32
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x10} ---
+++ killed by SIGSEGV +++
Segmentation fault

```

</details>

ferboiar, 2017-07-20:
hi guys!

meanwhile I am mounting my windows shares through fstab and link the directories as "Local" from nextcloud external storage .

But I don't want to touch operating system config files, I don't know if future nexcloudpi updates will overwrite manual file changes...

So, is there a workaround to config smb directly from nexcloud external storage?

regards ! and very good job nacho!! I'm very happy with your adaptation.

nachoparker, 2017-07-21:
Hi,

I will never modify system files in a way that overrides manual modifications. At least not automatically.

Raspbian Stretch is about to come out, so I expect this issue to be solved upstream. I will re-test this whenever I migrate NCP to stretch.

It is expected to come out this summer!

In the mean time, you would have to use a mount command into the data folder if you want to have an external SMB share visible in your Nextcloud interface.

IDK how that would affect sync though

jb8628, 2017-07-27:
I was able to get SMB working through External Storage after installing php-smbclient from **stretch**. However, performance wasn't good. I haven't investigated further as it's easier for me to ditch SMB altogether. There is also php-smb in **stretch**, which I haven't played around with.

nachoparker, 2017-07-28:
good one! I had tried installing smbclient from stretch but not that. I'll include that in the new image. Meanwhile, anybody can

```
sudo apt install -t stretch php-smbclient
```
closing now

nachoparker, 2017-08-08:
Ok, as it stands, it is either samba or `smbclient` ( see #82 )

I will revert this and `php-smbclient` will **NOT** be included in the images for now.

If anybody wants it ( for the External Storage App ), then install it manually, knowing that it will break samba.

nachoparker, 2017-08-17:
good news everyone!

Raspbian Stretch is here! I will upgrade NCP soon

https://www.raspberrypi.org/blog/raspbian-stretch/

nachoparker, 2017-08-20:
I can confirm that this has been fixed with the upgrade to Raspbian Stretch

closed 2017-08-20 by nachoparker