nextcloud/nextcloudpi #49: Prompted to upgrade v0.13.0 to v0.16.2 - tries but fails

Jakethethird, 2017-07-09:
Hi,

(See cut and paste from terminal in attached file)

1. I SSH in and am told I'm on v0.13.0
2. sudo nextcloudpi-config
3. Prompted immediately to upgrade to v0.16.2 (see screenshot)
4. Accept yes
5. Downloads nc-format-USB & nc-forward-ports
6. Says I'm on v0.16.2
7. sudo nextcloudpi-config
8. Buggy interface appears with fail2ban and another only partially present on the menu (see end of text file for cut and past of the config menu)
9. Restart
10. Back on v0.13.0 when it starts. 
11. Repeated 3x

Best regards,
Jake

[image]
[20170709 nextcloudpi bug.txt](https://github.com/nextcloud/nextcloudpi/files/1134026/20170709.nextcloudpi.bug.txt)

nachoparker, 2017-07-10:
step 9, do you mean _reboot_?

also, what changes did you do to the image? it seems like it is looking for some french locales that aren't there

Jakethethird, 2017-07-10:
Hi,
This installation has been working fine for several weeks
- Timezone was set to Paris (via nextcloudpi-config) after original installation, language kept as default.
- Yes, step 9 is sudo reboot.
- Step 11 should be "reproduced 3 x" (for testing if this is reproducible) - each time same result.
- I've just noticed that after this unsuccessful upgrade, I'm prompted to upgraded 28 packages. I already did this last night, now being asked to do it again - same packages (see file).

Jake

[20170710a nextcloudpi bug.txt](https://github.com/nextcloud/nextcloudpi/files/1134478/20170710a.nextcloudpi.bug.txt)

nachoparker, 2017-07-10:
so weird

Is your `nextcloudpi-config` also corrupt if you use the `v0.13.0` version? ( answering 'no' to the upgrade prompt )

The fact that Raspbian doesn't seem to write consistently to your SD card is starting to make me think that you could have a damaged SD card. Can you try this by copying the SD card to your PC and back?

If there are I/O errors then that is the cause

Jakethethird, 2017-07-10:
Hello,
You identified the problem. Fortunately I imaged my SD card 2 weeks ago, so restored the image onto another card. This booted to v.0.13.0, upgraded fine to v.0.16.2. Raspbian updated fine as well and all changes were maintained after a reboot.

To test the original card, I mounted it on my linux PC and created new files in both existing boot and ext4 partitions. These looked OK, but when I unmounted the card, then remounted it, they were gone. Thus the card seems to have failed to a read-only status where all subsequent changes were held (in ram? cache?) but then disappeared upon reboot (including data created in the last week in NC). This also explains why each time I rebooted, the NC sync clients would no longer connect and I had to regenerate new device passwords that worked until the RPi rebooted (and reverted to the previous ones), and why all the last week's notes either disappeared, or reverted to earlier versions.

However... one of the minor issues identified is still present and is unrelated to this card failure. It is the incomplete menu in the nextcloudpi-config menu after upgrade to v.0.16.2. See uploaded file.

Thanks for you attention and sorry that a card failure wasted your time. Really nice project.

Jake

[image]

Jakethethird, 2017-07-10:
And I think I've resolved the missing menu item issue by chance.
I looked up the reason for the language issue resulting in this warning comment during NC upgrade 
"Can't set locale; make sure $LC_* and $LANG are correct!"

Surprisingly it is related to the [SSH client](https://www.raspberrypi.org/forums/viewtopic.php?f=50&t=11870), specifically this comment:

_If it happens when using SSH, this is a fault on the SSH client, not the RPi. For example using an Ubuntu machine as the SSH client will cause this problem.
To fix this SSH problem, edit the file /etc/ssh/ssh_config on the SSH client (not the RPi) and remove the line
SendEnv LANG LC_*
_This stops the client sending invalid locale information to the RPi.__

When I do this it solves the language issue, but also results in the 6 missing menu items reappearing.

I think you can close this now!

Thanks again,
Jake

nachoparker, 2017-07-10:
Nice, I am glad we found the issue!

Also, good detective work with the locale ;)

closed 2017-07-10 by nachoparker