nextcloud/nextcloudpi #51: Collabora or onlyoffice

paarsand, 2017-07-10:
Please add functionality for one of these. Thats the only feature missing right now.

paarsand, 2017-07-10:
The image from techandme has this, but since your image is become oficial, will their image co-exist? Or will your's take over?

nachoparker, 2017-07-11:
hi, this has been requested a number of times, so I want to look at it.

It is not as easy though, because the Raspberry Pi  is an ARM architecture and to my knowledge the collabora server runs on x86. 

AFAIK they are in charge of the x86 virtual machine. This is the Raspberry Pi image.

paarsand, 2017-07-11:
Take a look at this: https://github.com/techandme/NextBerry

Its running raspbian lite.

nachoparker, 2017-07-11:
mmm but are you talking about the Nextcloud App or the server?

AFAIK the server is only available on x86. A quick look at their code shows that they are pulling the x86 docker from dockerhub ( [link](https://hub.docker.com/r/collabora/code/) )

tflidd, 2017-11-04:
Online office applications are already painfully slow but on a raspberry pi? Why do I want that? From my point of view it is a challenging task to make a RPi become a decent cloud storage and I'd focus on that.

nachoparker, 2017-11-05:
@tflidd  

Well, we are looking also at other more powerful ARM boards, but it is true that with the resources we have, this remains a low priority

enoch85, 2018-02-16:
@paarsand Thanks for pointing that out. @ezraholm50 is the responsible dev on that image, and he's been away for a while so it hasn't been updated. Though I think he will continue his work soon.

The NextBerry is a "clone" of the official VM but adopted for Raspberry Pi 3 (and 2 I think).

closed 2018-05-04 by nachoparker