nextcloud/nextcloudpi #53: Continuing tar.bz2 Issue

cottoa00, 2017-07-14:
I'm at a loss... 

The reason I checked the md5sum is because after I
downloaded and untarred the tar.bz2, I burnt the image on a fresh
SanDisk Ultra 32G card using the;

sudo dd bs=4M if=NextCloudPi_06-28-17.img of=/dev/sda1

Inserting into a new Pi3 and applying power... no boot sequence or
flashing light on board :-(... had monitor and keyboard attached.

Thought maybe had a bad card so repeated process with a fresh SD (same
spec)... still no boot?

Then repeated download, untar, burn (onto reformatted SD #2)... again
the same. I have another fresh Pi3 but don't think that is the issue...
any help appreciated.

cottoa00, 2017-07-14:
Just did process over again with new SD card, new card reader, new Pi3...

Here is what sequence looked like;

pi@jessie-hs3-nr:~/Downloads $ md5sum NextCloudPi_06-28-17.tar.bz2
2217552d5d33b927e3c85a5c5953984b  NextCloudPi_06-28-17.tar.bz2

pi@jessie-hs3-nr:~/Downloads $ tar -xvf NextCloudPi_06-28-17.tar.bz2
NextCloudPi_06-28-17.img
pi@jessie-hs3-nr:~/Downloads $ df -h
Filesystem             Size  Used Avail Use% Mounted on
/dev/root               29G   19G  8.9G  68% /
devtmpfs               426M     0  426M   0% /dev
tmpfs                  430M   19M  412M   5% /dev/shm
tmpfs                  430M  6.3M  424M   2% /run
tmpfs                  5.0M  4.0K  5.0M   1% /run/lock
tmpfs                  430M     0  430M   0% /sys/fs/cgroup
//192.168.1.103/share  1.8T  138G  1.7T   8% /home/pi/buffy
/dev/mmcblk0p1          63M   21M   42M  34% /boot
tmpfs                   86M  4.0K   86M   1% /run/user/1000
/dev/sda1               30G   32K   30G   1% /media/pi/174D-4E13
pi@jessie-hs3-nr:~/Downloads $

pi@jessie-hs3-nr:~/Downloads $ sudo dd bs=4M if=NextCloudPi_06-28-17.img of=/dev/sda1
567+1 records in
567+1 records out
2378414080 bytes (2.4 GB) copied, 198.782 s, 12.0 MB/s
pi@jessie-hs3-nr:~/Downloads $ df -h
Filesystem             Size  Used Avail Use% Mounted on
/dev/root               29G   19G  8.9G  68% /
devtmpfs               426M     0  426M   0% /dev
tmpfs                  430M   12M  419M   3% /dev/shm
tmpfs                  430M  6.3M  424M   2% /run
tmpfs                  5.0M  4.0K  5.0M   1% /run/lock
tmpfs                  430M     0  430M   0% /sys/fs/cgroup
//192.168.1.103/share  1.8T  138G  1.7T   8% /home/pi/buffy
/dev/mmcblk0p1          63M   21M   42M  34% /boot
tmpfs                   86M  4.0K   86M   1% /run/user/1000
/dev/sda1               30G   32K   30G   1% /media/pi/174D-4E13
pi@jessie-hs3-nr:~/Downloads $

Can't understand why final df -h shows no 2.4G copied to card?!
Know this is probably not your issue, but with all new cards, readers, etc not sure what else I can do on my end? Yikes!

cottoa00, 2017-07-14:
Update... pushed .img to my windows drive and used SDFormatter to burn SD... this time it worked!? No clue what happened with dd (several times on several SDs)... whole Pi thing is to move away from Windows ;-( 

BUT... it is working... sorry for the confusion.

closed 2017-07-14 by cottoa00

nachoparker, 2017-07-14:
glad you got it working. 

For the record, you have to use the block device, not the partition in `dd` like so

```
sudo dd bs=4M if=NextCloudPi_06-28-17.img of=/dev/sda  status=progress && sync
```
Also, you have to sync at the end