nextcloud/nextcloudpi #54: Overwrite cli url

deljones, 2017-07-16:
Just want to check something here.. 

Can I change the overwrite cli url in config.php to point to my domain rather than http://localhost.?

The reason I ask is that the small amount of users on my system, when they get activity notifications the notification has a link to http://localhost which of course when they click on it goes no where... 

Just checking I can do this without blowing up my perfectly working ND install

Del

nachoparker, 2017-07-18:
AFAIK, the cli URL parameter is meant for systems behind a proxy to know what the real URL is.

See [this](https://docs.nextcloud.com/server/12/admin_manual/configuration_server/config_sample_php_parameters.html#proxy-configurations)

I haven't investigated where those notification URLs come from, a quick search points in the direction you went

https://help.nextcloud.com/t/localhost-in-activity-notifications/12753

If you want to experiment with that parameter, it's fine. Just make a backup copy of `config.php` and you should be fine. Just don't touch other parameters ;)

Let us know what you discover!

deljones, 2017-07-18:
So I changed the overwrite cli url in the config to reflect the domain that the instance is sitting in. Now of course when users get their activity report the http://localhost is now replaced with the domain address allowing users to login and see their activity. I read the link about overwrite cli url but its still a bit vague. Changing it in the config file has made no difference to my NC install... Whew! but now users have a meaningful link to go to.

I looked at a long running ownCloud 9 instance that I have had running for a while and the overwrite cli url is already filled in with the correct domain.

Not sure if this helps but that's what I've found thus far..

D

nachoparker, 2017-07-18:
good job, I'll try it and include it then. I'll make it so whenever you setup a domain through dnsmasq or letsencrpyt, that value get updated

thanks!

ferboiar, 2017-07-19:
Today I configured let'sencrypt and because of that I had to create a free domain in freenom pointing, with cname, to my dynamic dns (letsencrypy told me that there were too many requests for dtdns.net).

When doing this the images have stopped loading, with the help of https://community.letsencrypt.org/t/cname-should-point-to-dynamic-dns-service/14223/2 I modified config.php like this:

'Overwrite.cli.url' => 'https://cloud.myserver.com',
'Overwriteprotocol' => 'https',

This way the web is accessible externally without certificate errors or problems loading the images.

I expose it here because it is related to what says @deljones and should be contemplated in future versions of nextcloudpi :)

nachoparker, 2017-07-19:
@ferboiar 

Hi, thanks very much for the contribution.

I have been looking at those parameters in the past but like I said, I concluded that were related to proxy setups only... in any case, I guess t will not harm to set them automatically, so I will include that.

Many thanks

;)

nachoparker, 2017-07-28:
implemented in v0.17.17

closed 2017-07-28 by nachoparker