nextcloud/nextcloudpi #55: nc-init takes forever

JVAQUEROM, 2017-07-17:
Hello, I've had the problem that the webpage shows blank after a reboot (error 500 on Chromium). I tried moving the nc-database somewhere and then back to /etc/lib/mysql, but no luck. MariaDB seems to fail always. So I decided to do a "nc-init" to restore everything and see if that way it might work. The screen shows:

`Launching nc-init
Setting up database...
`

How much time is it supposed to take? It been more than half an hour... I think something is wrong.

nachoparker, 2017-07-18:
If you tried to move the database back to the SD, I assume that you already read #15 

I guess you have tried `mpgrep mysql` and the output is empty.

The reason `nc-init` takes forever is that it waits for mysql to be up ( as you can see [here](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-init.sh#L44) )

So let's see, the location of the configuration has changed, so depending on what image you installed in the first place, you want to look at
 - `/etc/mysql/my.cnf`, or 
 - `/etc/mysql/conf.d/50-server.cnf`

In that file, look for the `datadir` parameter, and make sure it is `/var/lib/mysql`.

 - make sure that that folder exists and is owned by the user `mysql`
 - try to start the database `systemctl restart mysqld`, and then look at the logs
   - `systemctl status mysqld`
   - `tail /var/log/mysql.log`

Thanks for the feedback, I will include some sort of timeout in `nc-init` for the case where the database is down.

JVAQUEROM, 2017-07-18:
Yes, I read [#15](https://github.com/nextcloud/nextcloudpi/issues/15).

About `mpgrep mysql`, my pi says "command not found".

I installed from the ISO in ownyourbits page. There is nothing about datadir in /etc/mysql/my.cnf and there is no file /etc/mysql/conf.d/50-server.cnf. However, it does exist /etc/mysql/mariadb.conf.d/50-server.cnf with the line `/etc/mysql/mariadb.conf.d/50-server.cnf`. Should I move it?

The mysql folder is owned by `mysql`

I'll post know the outputs of the commands about mysql. I did not have a `/var/log/mysql.log` but a `/var/log/mysql/error.log`. Here it  goes:

`root@rasppberrypi:/home/pi# systemctl restart mysqld
Job for mariadb.service failed. See 'systemctl status mariadb.service' and 'journalctl -xn' for details.
root@rasppberrypi:/home/pi# systemctl status mysqld
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled)
  Drop-In: /run/systemd/generator/mysql.service.d
           └─50-mariadb-$database.conf
   Active: failed (Result: exit-code) since Tue 2017-07-18 19:05:02 CEST; 4s ago
  Process: 8078 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
  Process: 7980 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 7975 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 7973 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 8078 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Jul 18 19:04:59 rasppberrypi mysqld[8078]: 2017-07-18 19:04:59 1988792320 [Note] /usr/sbin/mysqld (mysqld... ...
Jul 18 19:05:02 rasppberrypi systemd[1]: mariadb.service: main process exited, code=exited, status=1/FAILURE
Jul 18 19:05:02 rasppberrypi systemd[1]: Failed to start MariaDB database server.
Jul 18 19:05:02 rasppberrypi systemd[1]: Unit mariadb.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.
root@rasppberrypi:/home/pi# tail /var/log/mysql.log
tail: cannot open ‘/var/log/mysql.log’ for reading: No such file or directory
root@rasppberrypi:/home/pi# tail /var/log/mysql/error.log
2017-07-18 19:04:59 1988792320 [Note] InnoDB: Highest supported file format is Barracuda.
2017-07-18 19:04:59 1988792320 [Note] InnoDB: 128 rollback segment(s) are active.
2017-07-18 19:04:59 1988792320 [Note] InnoDB: Waiting for purge to start
2017-07-18 19:05:00 1988792320 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3105611
2017-07-18 19:05:00 1421308912 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-07-18 19:05:00 1988792320 [Note] Plugin 'FEEDBACK' is disabled.
2017-07-18 19:05:00 1988792320 [Note] Recovering after a crash using tc.log
2017-07-18 19:05:00 1988792320 [ERROR] Can't init tc log
2017-07-18 19:05:00 1988792320 [ERROR] Aborting`

Thank you!!

nachoparker, 2017-07-18:
Sorry, I mistyped (and corrected) the command, its `pgrep mysql`. Also I didn't have access to a new version at that moment, so I was typing by memory.

Find the line for `datadir` in `/etc/mysql/mariadb.conf.d/50-server.cnf`, and make it into

```
datadir = /var/log/mysql
```
Then issue
`sudo systemctl restart mysqld`

And read the logs, `pgrep` and `status` command again. If it is still down with the same message

```
Can't init tc log
```

Try [this](https://bbs.archlinux.org/viewtopic.php?id=206379)

JVAQUEROM, 2017-07-18:
As you said, `pgrep mysql`output is empty. 

I have set datadir to /var/log/mysql and now the output error is different:

`2017-07-18 19:50:22 1989263360 [Note] InnoDB: Creating tablespace and datafile system tables.
2017-07-18 19:50:22 1989263360 [Note] InnoDB: Tablespace and datafile system tables created.
2017-07-18 19:50:22 1989263360 [Note] InnoDB: Waiting for purge to start
2017-07-18 19:50:22 1989263360 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 0
2017-07-18 19:50:24 1413477360 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-07-18 19:50:24 1989263360 [Note] Plugin 'FEEDBACK' is disabled.
2017-07-18 19:50:24 1989263360 [ERROR] Could not open mysql.plugin table. Some plugins may be not loaded
2017-07-18 19:50:24 1989263360 [ERROR] Can't open and lock privilege tables: Table 'mysql.servers' doesn't exist
2017-07-18 19:50:24 1989263360 [Note] Server socket created on IP: '127.0.0.1'.
2017-07-18 19:50:24 1989263360 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'mysql.user' doesn't exist`

nachoparker, 2017-07-18:
it should be `/var/lib/mysql`

JVAQUEROM, 2017-07-18:
It worked with the fix your gave! removing /var/lib/mysql/tc.log did the magic.

However, I seem to have a permissions problem with my hard drive where the nextcloud data is stored. How should I mount it? What permissions/owner should ncdata have? RIght now, root is the owner..

P.D.: owyourbits webpage seems to be down, I cannot access...

nachoparker, 2017-07-18:
That's great! :)
Please, open a different issue for each separate technical question, so users can find the solutions easily.

PD: I know about the website... not happy with the new hosting at all, sorry about that :(

nachoparker, 2017-07-27:
improvement implemented in v0.17.7 ae7a979

closed 2017-07-27 by nachoparker