nextcloud/nextcloudpi #56: Problem when switching from NTFS to ext4

eogut, 2017-07-17:
Hi,

I changed the filesystem of my external from NTFS to ext4, which caused a bootloop on my Raspi because I didn't remove the respective line from fstab first. Good thing it booted into recovery mode so I did that. While I was at it, I changed the swapfile configuration, as it was located on the external. In order to do that, I just removed both the location and the filesize... Now I can boot into the system but when I'm in nextcloudpi-config, I get the error that there's no space left on the device, so I can't relocate the database to my external in order to pick up where I left off. If I plug-in the SD card into my Windows computer, it shows me that there's 19,4 MB of 40,3 MB left... What should I do?

Thanks!

nachoparker, 2017-07-18:
Did you set the new ext4 partition in `/etc/fstab`? Keep in mind that if automount and moving the swap don't play nice together at the moment.

Look at #37

In any case...

 - Log in
 - `swapon -s`, to discover where your swapfile is
 - umount your drive `sudo umount /media/USBdrive`
 - `sudo swapoff /route/toyour/swapfile`
 - `sudo rm /route/toyour/swapfile`
 - either
    - place in `/etc/fstab` your new `ext4` partition, reboot, move the swap again
    - move your swap back to SD card

hope it helps. I'll close but we can continue talking

nachoparker, 2017-07-18:
Duplicate of #37

closed 2017-07-18 by nachoparker

eogut, 2017-07-23:
Thanks for your help! The swapfile command didn't come up with anything so I just disabled it using `sudo systemctl disable dphys-swapfile`, but the problem persists. So I tried to use `sudo apt-get purge dphys-swapfile` but terminal returned: 
`Reading package lists... Error!
E: Write error - write (28: No space left on device)
E: Can't mmap an empty file
E: Failed to truncate file - ftruncate (9: Bad file descriptor)
E: The package lists or status file could not be parsed or opened.`

nachoparker, 2017-07-25:
mmm you need to free up some space before fixing. Let's try step by step

first, these commands should give you a little room to work with
```
apt-get autoremove -y
apt-get clean
rm /var/lib/apt/lists/* -r
```
Delete something else if you can think of something that is big.

Then, let's see, what's the output of
```
swapon -s
```
```
ls -lh /var/swap
```