nextcloud/nextcloudpi #57: permissions on ncdata

JVAQUEROM, 2017-07-18:
I had a problem with the data folder, ncdata, permissions. It turns out the folder must be owned by `www-data`. I was wondering if it is possible to make `nc-datadir` in `nextcloudpi-config` change automatically the owner to `www-data`.

It is not a big deal but it took me some time to realize what was the proper owner. I guess that there may be somebody even "newbier" than me that could have some bad time with this little issue. 

Just a suggestion. Keep up the good work. Nextcloudpi is AWESOME!!

nachoparker, 2017-07-18:
thanks

`nc-datadir` copies the original datadir ( [code](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-datadir.sh#L62) ), so the owner doesn't change. 

Also, it checks that the file system supports permissions.

Did you copy the directory around manually? How did this situation occur?

JVAQUEROM, 2017-07-18:
I believe it is due to some issues with automount. My hdd is divided in two partitions, and I do not really know what happened but in the end I edited `/etc/fstab` so that the partitions were mounted in the original place always. Maybe it is due to that, but I don't know for sure.

nachoparker, 2017-07-19:
Mmm, okay

As I wrote in the [instructions](https://ownyourbits.com/2017/05/24/nextcloudpi-gets-automount-backups-nextcloud-online-installation-nextcloud-12-and-more/), it is ideal to use the label of the partition.

I have tested with partitions and it worked fine for me. In any case, I will re-test that just in case.

Maybe I should just only automount the partition if it has a label...

If anybody else has run into this, any info would be great

closed 2017-07-19 by nachoparker