nextcloud/nextcloudpi #58: Plex + NextcloudPI

zahareevici, 2017-07-19:
Have somebody installed plex in nextcloudpi ? 
installation was without any problems but is not starting .

Please help

nachoparker, 2017-07-19:
Hi, 

I have not tried that myself, but it would probably be useful to provide logs and a description of what is going on.

Also, as soon as Raspbian 9 stretch comes out, I will migrate NextCloudPi so potential problems coming from using the stretch repo for Apache and Mariadb should be solved.

JVAQUEROM, 2017-07-19:
maybe you have to go to `raspi-config` to change the memory split between the system and the graphical card.

The 1Gb RAM of raspberrypi is both for the standard ram and the GPU RAM. By default, it is just 64Mb, probably not enough for streaming video. Something similar happened to me when installing Kodi, I had to change the GPU memory from 64 MB to 256 MB.

I am no expert in the matter, this is just a user commenting!! I just hope its helpfull!!

zahareevici, 2017-07-20:
the problem is that plex is not starting at all, with clean raspbian and php 7, mariadb al is working good.

zahareevici, 2017-07-22:
@nachoparker  is there an instruction maybe how to install NextCloudPi in raspbian ? with just `installer.sh` i get the error :(

zahareevici, 2017-07-23:
fixed, first install plex, after configure NextcloudPi, upgrades, etc.

nachoparker, 2017-07-24:
I am glad. Other people trying the same thing will have a solution here :D

closed 2017-07-24 by nachoparker