nextcloud/nextcloudpi #59: Cannot lock in to users after sharing

new-on-github, 2017-07-20:
I have a Problem:

Situation: I create a new user, lock the new user in, everything seems fine.

Problem: After sharing something with the new user, the lock-in doesn't work any more. The web gui says

> Service Unavailable
The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.
Apache/2.4.25 (Raspbian) Server at [...]

new-on-github, 2017-07-20:
Sorry, was my fault!

nachoparker, 2017-07-20:
haha thanks, I'll test this soon

new-on-github, 2017-07-20:
If I share something from a "normal"-user to the admin-account, the log in to the admin-account also fails.
--> Sharing seems to be impossible...

new-on-github, 2017-07-24:
Any news on this problem?

nachoparker, 2017-07-25:
Hi,

I have tested manually with a fresh installation and could not reproduce this problem. Steps I took

 - log in admin user
 - create test user
 - upload a small file
 - share with test user
 - log in test user -> ok
 - upload a small file
 - share with admin user
 - logout and in with admin user -> ok

closed 2017-07-25 by nachoparker