nextcloud/nextcloudpi #60: Cannot update NexCloudPi - No Internet Connectivity

jcaraba, 2017-07-20:
root@nextcloudpi:/usr/local/bin# ncp-test-updates
No internet connectivity
root@nextcloudpi:/usr/local/bin# ncp-update
No internet connectivity
root@nextcloudpi:/usr/local/bin# ncp-check-updates
No internet connectivity

NextCloudPi v0.12.6 is up to date

root@nextcloudpi:/usr/local/bin# nextcloudpi-config
No internet connectivity
No internet connectivity
Launching nc-update
No internet connectivity

nachoparker, 2017-07-21:
Might sound like a silly question but... do you have internet connectivity?

Can you `ping ownyourbits.com`?

jcaraba, 2017-07-21:
I take it back, I can't ping anything outside of my network. I can however
ping anything inside, ssh and connect to NextCloudPi from outside my
network.

I'm investigating...

closed 2017-07-21 by nachoparker