nextcloud/nextcloudpi #61: Problem with forwarding ports

JVAQUEROM, 2017-07-22:
Hi!

When I followed the guide to set newxtcloudpi, I jumped the forwarding ports and no-ip because first I wanted to try inside my network before open my pi to the world. Now, I followed the steps but when I try to access my nextcloud from outside I get my router configuration portal (the one to change the UPNP and so on, asking for a sign-in).

The upnp forwarding ports in nexctcloud-config seems to forward the external port  443 and the internal port in my pi 443. I do not know if this is correct, but I would guess that it should be the port 80 in my pi...

This is the result of nc-forward-ports:
`Found valid IGD : http://192.168.1.1:5431/uuid:f0041001-5c00-0084-6cf0-005c02109078/WANPPPConnection:1
Local LAN ip address : 192.168.1.***
UPNP_DeletePortMapping() returned : 0
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.1.1:5431/dyndev/uuid:f0041001-5c00-0084-6cf0-005c0010ec10
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

Found valid IGD : http://192.168.1.1:5431/uuid:f0041001-5c00-0084-6cf0-005c02109078/WANPPPConnection:1
Local LAN ip address : 192.168.1.****
ExternalIPAddress = xx.xxx.xx.xx
InternalIP:Port = 192.168.1.***:443
external xx.xxx.xx.xx:443 TCP is redirected to internal 192.168.1.****:443 (duration=0)
Done. Press any key...`

nachoparker, 2017-07-24:
The output looks good to me. 

Port 80 is for the unsecure and unencrypted HTTP protocol
Port 443 is for the secure and encrypted HTTPS protocol

It is only recommended to use HTTPS, specially if you plan to access from outside your LAN. That is why is the only option that I made available in `nextcloudpi-config`.

Just access through HTTPS, it seems like the forwarding configuration succeeded

closed 2017-07-24 by nachoparker

JVAQUEROM, 2017-07-24:
You are right, accessing through https solves the issue.

Big thanks!!!

Closed #61<https://github.com/nextcloud/nextcloudpi/issues/61>.