nextcloud/nextcloudpi #62: Problem with DDNS (No-IP)

jinxoO, 2017-07-23:
Hi, 
first thanks for the great work and your effort! Nextcloudpi is really a great project!

Now the Problem: 
when i try to set up the integrated no-ip service i always get the following error message, though username and password are correct:
`Launching no-ip

Auto configuration for Linux client of no-ip.com.

You have entered an incorrect username
	-or-
an incorrect password for this username.

System config value trusted_domains => 3 set to string MY-NO-IP-DOMAIN.ddns.net
Done. Press any key...`

When i check the status with sudo no-ip2 -S i get 
`sudo noip2 -S
 Can't locate configuration file /usr/local/etc/no-ip2.conf. (Try -c). Ending!`

I've installed version v0.16.2 and actually everything works fine till i get a new public-ip from my isp (obviously)

Any suggestions?

nachoparker, 2017-07-24:
Just to make sure... have you set up a free account with noip? does your RPi have internet access?

jinxoO, 2017-07-24:
You are right! I have set up a free account and the RPi has internet access. I can access NC from "outside" with the registered No-IP-Domain. But only for perhaps one day, till my ISP changes the public-IP. When I refresh the IP manually via the No-IP Website everything works again (for a day). The problem is, I can not set up the No-IP-client correctly.

nachoparker, 2017-07-25:
The only thing you have to do is to use your registered domain and password in `nextcloudpi-config` (or the new web interface :) )

Are you 100% sure that you are using your noip credentials, and _ACTIVE=yes_?

nachoparker, 2017-07-25:
I just tested this, and it works for me. 

USER: email that I used to register at noip.org
PASS: the password that I used for noip.org
DOMAIN: the domain that I registered at noip.org

Triple check the data. If anybody else runs into this, please reopen. 

@jinxoO I'll close, but we can continue talking

closed 2017-07-25 by nachoparker

jinxoO, 2017-07-26:
Ok, thanks!  It works for me now, although I don't know what the problem was (still is?)?!

Everytime I try to enter USER, PASS and DOMAIN in <code>nextcloudpi-config</code> I get the error mentioned in the first post. 

I tried creating the config file manually with <code>sudo noip2 -C</code> and got no error. But still the public ip did not refresh automatically. 

When I activated the noip client in <code>nextcloudpi-config</code> to <code>ACTIVE=yes</code>, without changing anything else, everything worked.  

I dont know why <code>nextcloudpi-config</code> does not work for me. I tried to enter my noip credentials a few times after this solution (again) without any success. But I think it is a negligible problem 😉

nachoparker, 2017-07-26:
awesome, I am glad

GustaveCoste, 2017-07-26:
I seem to have exactly the same issue. I will try to edit the config file manually as @jinxoO did.

nachoparker, 2017-07-26:
@GustaveCoste thanks, 

You should not need to manually edit anything. Try through `nextcloudpi-config` or the web interface. You should be able to ;)

GustaveCoste, 2017-07-28:
I also solved the problem with `sudo noip2 -C`.
I'm 200% sure I was using the good credentials in nextcloudpi-config or in the web interface but it always told me the same error about incorrect password/username.

Now it seems to be working.

nachoparker, 2017-07-29:
retested... works for me... anybody else seeing this?

rlueder, 2017-12-30:
I'm having the same issue as @jinxoO and @GustaveCoste, I signed up for a paid acct. with No-IP and I know I'm entering the right information because my router can access it but I keep getting the same error from the first post in this issue.

seekheart, 2018-02-19:
I'm on a raspberry pi 3 and when I tried installing the Duc it errors stating that my password/username is incorrect. I'm able to login just fine on the website with the same credentials and I even changed the password to something far easier to type as a sanity check. Both cases have led to the same error... How to fix...

kozbulski, 2019-01-02:
Hi,
maybe this helps:
I had the same problems. I also use this information:
-> USER: email that I used to register at noip.org
-> PASS: the password that I used for noip.org
-> DOMAIN: the domain that I registered at noip.org
but it only worked after i changed my password: it now contains one number a . and capital letters.
It didn't work with a ! as last character.

Can't say if its my nextcloud or no-ip.
I was able to log into no-ip with my old password. 

NextCloudPi version | v0.67.11
NextCloudPi image | NextCloudPi_02-06-18
Nextcloud version | 14.0.4.2

Regards

irishgeoffrey, 2019-04-04:
you can also use https://staticIP.io if you need a static IP

Skull40, 2020-04-18:
Forgot password from noip.com
Changed the password through the password recovery system, the password has earned a new password, but only to enter the site.
In the account settings section, the password was changed to the same :) and now the new password works everywhere.

VerstreuteSeele, 2021-04-09:
Got a similar issue to some of the people

When I restart my pi, noip doesnt seem to work correctly. It is running according to "sudo noip2" > noip is already running
but it doesnt work. I have a static ip so theres nothing changing however everytime my pi reboots I gotta go into the ncp-config and set up noip again... and im also getting the error that there is no config file for noip before I do that.. It is there tho and I can access it,

Any ideas?

NICKVALENTINO, 2023-05-05:
> 

Thanks a lot - this helped me out!