nextcloud/nextcloudpi #63: Cannot connect from my own network

JVAQUEROM, 2017-07-24:
Hi there!

I have my nexcloudpi up and running, but it seems that if I try to go to it (through the no-ip direction) when I am connected to the same network as my RPI, I get "I cannot connect" from my browser. Nextcloud client cannot connect as well.

nachoparker, 2017-07-25:
you probably need to set `dnsmasq` from `nextcloudpi-config` ( or the new web interface :) )

Read [this](https://ownyourbits.com/2017/03/09/dnsmasq-as-dns-cache-server-for-nextcloudpi-and-raspbian/)

closed 2017-07-25 by nachoparker

JVAQUEROM, 2017-07-26:
It worked, thank you. After  I while I could not connect to wifi for a long time, so I changed my router configuration (I had set first DNSserver as my pi's IP and second DNS server as the first in my router). I do not know maybe it was not related, but it seemed so...

But configuring only the devices I am going to use in my network works as well.