nextcloud/nextcloudpi #65: config.php

fzrkid, 2017-07-26:
Can you add the ability to edit config.php from the new web interface?

Just wanted to confirm, if the ncdata and ncdb are moved to the USB drive /var/www remains on the sdcard correct. Apache root remains on the SD card.

Is the config.php where we set new db user/password

nachoparker, 2017-07-26:
Good idea! 

> Just wanted to confirm, if the ncdata and ncdb are moved to the USB drive /var/www remains on the sdcard correct, so Apache root remains on the SD card?

That's correct

> Is the config.php where we set new db user/password

The db password needs to be set in the database (in mariaDB), but then it has to be reflected on `config.php` so NC can access the database.

nachoparker, 2018-04-03:
implemented in 'v0.53.12', at least we can see it, might improve it so we can actually edit it from here, but that requires some thought, as things can break badly and it's better not to touch this file if you don't know what you are doing.

thanks

closed 2018-04-03 by nachoparker