nextcloud/nextcloudpi #66: Ftp server to nextcloud

polmourgos, 2017-07-26:
I have just installed the latest version to my RPi3.I have everything and works like it shoul be.Very good and great job.
My right now setup is dietpi with nextcloud on Odroid XU4.
But I wanted to give a try for some period with rpi3 so I started.
I have some questions and problems!!!!
I cant login with putty from my laptop.I dont know if its possible and if yes i dont know how to do it.
And is it posible to run a ftp server to the nextcloudpi?
Thank you for you job and your time.
Pol

JimmyKater, 2017-07-26:
> I cant login with putty from my laptop.I dont know if its possible
should be possible
> and if yes i dont know how to do it.
open putty. enter your local ip-address (or nextcloud.local), make sure that "port" is set to 22
mark "SSH" in the checkbox "Connection type". and hit enter or press the "open"-box.
most probably putty will grumble that the SSL-certificate isnt know yet - tell putty to accept it since it's in your local netwerk.
usually your pi would ask you to enter a username (standard-user is pi) and a password (raspberry) and in you are. then type sudo raspi-config
hope that answered your questions and helped a bit.

nachoparker, 2017-07-26:
> I cant login with putty from my laptop.I dont know if its possible and if yes i dont know how to do it.

SSH is disabled by default, see #8 

> And is it posible to run a ftp server to the nextcloudpi?

Yes, if you install it yourself ;)

@JimmyKater thanks for the help! ;)

closed 2017-07-26 by nachoparker