nextcloud/nextcloudpi #67: can't connect to nextcloudpi server IP

Ricardosgeral, 2017-07-26:
Hi
I've facing difficulties to install NC

These are the basic steps i've done
-used Etcher to copy the image to microSd put it in the PI 3 model B and connect to power
- at this point i can't enter https://192.168.1.2 in the browser (it says it's unable to connect)
-entered https://192.168.1.2:4443 in chrome browser (in windows) -> this takes a long time ... (it doesn't seens normal to me)
- nc-format-USB
- nc-automount
- nc-init

these 3 last actions took a lot of time...

then 
- noip
- letsencript

After all these actions i try to enter https://xxxxx.ddns.net and what i received this

> Service Unavailable
> 
> The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.
> 
> Apache/2.4.25 (Raspbian) Server at xxxxx.ddns.net Port 443

I've opened the ports 443 and 80 in my router for the noip host

What i'am doing wrong? there a problem with this image of nextcloudpi? 

help is appreciated
cheers

nachoparker, 2017-07-27:
Hi,

First, let's make sure that you have the right IP ( `192.168.1.2` ). 

Assuming you have a keyboard and screen connected, could you paste the output of

```
ip address | grep global
```
?

The web panel on port 4443 should load instantly. Nextcloud does take a while, specially the first time you run it.

When stuff takes a long time and there is random failures and erratic behaviour, it is normally related to hardware problems.

 - Power supply: cheap power supplies can be unstable and affect performance. Is it a new PSU?
 - SD card: they can be easily corrupted (specially if you turn off the power without cleanly shutting down the system). You can try
    - `sudo halt`
    - insert the SD card in a PC and copy it into a file, in Linux is easy, I imagine you can also do this with echer... copying the other way. If it copies ok without I/O errors then at least read operations work.
    - re-copy that file into the SD card again, and see if there are I/O errors on write.

Hope it helps

Ricardosgeral, 2017-07-27:
Hi Nacho 
thanks for the help

I've installed everything from scratch...still no luck :(

ip address | grep global
returns
    inet 192.168.1.2/24 brd 192.168.1.255 scope global wlan0

after installing everything from start, now i can't even access https://192.168.1.2:4443 from home network

it results in chrome and firefox are something like

"This site can’t be reached
192.168.1.2 refused to connect.
Try:
Checking the connection
Checking the proxy and the firewall
ERR_CONNECTION_REFUSED"

the strange is that i can access via SSH to the no-ip host xxxx.ddns.net (even from outside my home network) 

I'm completely lost (maybe it's a basic thing but my experience is limited)

The nc-forward-ports script failed (i guess from the log - see attached file) 
[forwardports.txt](https://github.com/nextcloud/nextcloudpi/files/1181534/forwardports.txt)

I've opened all ports in the router 192.168.1.1 and enabled ddns. Can't find anything about UPnP in the router configurations.

About the SD card. I've tried already with 2 diferent SD cards  (16gb class 10). I will try to write a image of the card and do the reverse as you suggested. But i'm ceptical about being the SD card

About the power supply. I got an ultimate kit from Aliexpress that came with a 5V 3A power supply that doen't look cheap... I am using a USB 1T external 2,5" drive (with no power supply). Do you think i should consider a powered hub? from what i've seen the PI can work with one external disk without additional power. i'm using a minikeyboard (with mousse incorporated) with usb stick. 

Any other suggestions? i'm getting a little bit frustated ahah

thanks in advance

nachoparker, 2017-07-28:
Maybe your router doesn't have uPnP support. It doesn't matter, just do it manually then as you did.

Seems like the SSH daemon works, but either web server or database doesn't

I keep thinking that it's a hardware issue. Try with a fresh image, don't try to add the USB drive until you have everything working on the SD. If then adding the USB makes it fail, then power it externally.

Just try adding one thing at a time, and check each step ;)

Ali Express... Try another power supply if you can, even a 2.5A USB phone charger (I assume it's a RPi 3)

Ricardosgeral, 2017-07-29:
Hi Nacho

It was the f***ing cheap chinese powersupply. I bought a official RPI3  power supply and everything seems so far much more stable and fast.

You just saved me a lot of euros because i was considering getting a new external 1T external disk (to check if there was that the problem)

You just earn a beer from me :+1: 

How do i get rid of the "not secure" warning in red with the https with a line in chrome
I think that is why i don't have the nextcloud logo in the starting page (see image)
[image]

nachoparker, 2017-07-29:
nice! xD

I am glad we found it.

About the missing logo its a known bug with easy solution https://github.com/nextcloud/nextcloudpi/issues/35 (reboot)

About the security warning, use the `letsencrypt` extra. ( [info](https://ownyourbits.com/2017/03/17/lets-encrypt-installer-for-apache/) )

closed 2017-07-29 by nachoparker