nextcloud/nextcloudpi #68: USB drive partitioning / naming

fzrkid, 2017-07-27:
Using the new web interface and trying to understand how linux/nextcloud manages these things.

Why do I have 3 different directories on the USBdrive
/media
NCDrive_1 That is what I named the drive when I used the format tool
USBdrive I guess this is default
USBdrive1 not sure where that came from

I partitioned the drive with the web interface and named it NCDrive_1. Why doesnt the config.php show "/media/usbdrive/NCDrive_1/ncdata" or something to that effect to help with managing the USB drives?

Should I have used the NCDrive_1 name when moving the data and db directories?

The data directory is now /media/USBdrive/ncdata

ncbackup from the tool stored the backupfile in USBdrive

What I am trying to achieve is a dedicated drive for my NC server. Maybe have 1 directory for data and another for the DB. Nothing else anywhere.
But I always seem to end up with multiple 'mount points?' and the files are never where I want them.

The way I understand it from what I have read.
media is the physical drive
cd /media
Under that are 'mount points' example is USBdrive, USBdrive_1 or NCDrive_1
That would be:
cd /media/NCDrive_1
Now anything under those mount points are directories'folders' where the data is stored.
cd /media/USBDrive_1/ncdata

Whenever I look under any of the 'mount points' ie. NCDrive_1 or USBdrive I see the same files and folders.

Sorry for these newbie sort of questions. I have looked on-line and played around, but it still confuses me. I guess I am too accustomed to windows partitions.

I dont have any real data on NC yet, waiting until I fully understand all the workings so I can be relatively confident my data is safe. This way I can blow the thing out as many times as I need to get it right.

nachoparker, 2017-07-28:
Quoting [this](https://ownyourbits.com/2017/05/24/nextcloudpi-gets-automount-backups-nextcloud-online-installation-nextcloud-12-and-more/)

> Just activate it and your /media folder will be populated with an entry such as /media/USBdrive that will be already mounted for you for each connected USB drive. If the filesystem contains a label, a symbolic link will also appear in /media/.

```	
 $ ls -l /media
total 8
lrwxrwxrwx   1 root root   16 May 24 15:23 cloudDrive -> /media/USBdrive1
drwxrwxr-x+ 11 pi   pi   4096 May 24 15:23 USBdrive
drwxr-xr-x   4 root root 4096 Jan  1  1970 USBdrive1
```

> In this example, the first inserted USB drive does not have a label, but the second does, so a symbolic link appears to uniquely identify it.
If you want to use several USB mounts, you should reference the drive by its label in order to have a predictable path to it. This way, your configuration will not depend on the order in which USB sticks were plugged in.

So the way automount works is

 - a device named `/dev/USBdrive`, `/dev/USBdrive1`, and so on will be created
 - if automount is on, it will me mounted in `/media/USBdrive`, `/media/USBdrive1` and so on
 - if the device has a label, a _symbolic link_ will be also created pointing at `/media/USBdriveX` so you can reference it by label name

So in the above example, `/media/cloudDrive` is a _symbolic link_ to `/media/USBdrive1`. If possible, it is recommended to use the label, in case the order of plugging in things changes and what today is `/media/USBdrive1` tomorrow is `/media/USBdrive2`

Just use `/media/cloudDrive` in `nextcloudpi-config`

Now, `config.php` only manages Nextcloud. If you use `nc-datadir` to move the datadir, it will be reflected in `config.php` so that it knows where to look for the data dir

You can browse the [code](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-datadir.sh#L75) to see how it goes internally

So..

 - `/media/` is just a folder
 - I chose to create other folders inside named `/media/USBdriveX`
 - I use those as mountpoints, equivalent to doing `sudo mount /dev/USBdrive /media/USBdrive`
 - `/media/NCDrive_1` is a symlink to `/media/USBdriveX`

That explains why you get the same doing

```
ls /media/USBdrive
```

than doing

```
ls /media/NCDrive_1
```

Hope it makes sense ¯\(°_o)/¯

closed 2017-07-28 by nachoparker

fzrkid, 2017-07-28:
Thank you for taking the time to explain that. I will play around with the naming to ensure I have a thorough understanding.

nachoparker, 2017-07-28:
^^ sure

fzrkid, 2017-08-03:
I used the formatting tool in the web interface:
/dev/sda: 2 bytes were erased at offset 0x000001fe (dos): 55 aa

Welcome to fdisk (util-linux 2.25.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0x1c590331.

Command (m for help): Created a new DOS disklabel with disk identifier 0x52d488de.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-1953525163, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (2048-1953525163, default 1953525163): 
Created a new partition 1 of type 'Linux' and of size 931.5 GiB.

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.

Creating filesystem with 244190389 4k blocks and 61054976 inodes
Filesystem UUID: c8a34c58-1bb0-4a59-a7cf-de643370420e
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968, 
	102400000, 214990848

Allocating group tables:    0/7453#########         #########done                            
Writing inode tables:    0/7453#########         #########done                            
Creating journal (262144 blocks): done
Writing superblocks and filesystem accounting information:    0/7453######### 6/7453######### 730/7453#########         #########done

This is what I ended up with:
lrwxrwxrwx 1 root root   15 Aug  3 13:51 NCDrive_0 -> /media/USBdrive
drwxr-xr-x 3 root root 4096 Aug  3 13:50 USBdrive
drwxr-xr-x 3 root root 4096 Aug  3 13:50 USBdrive1

I used rmdir for USBdrive and USBdrive1 and now have:
lrwxrwxrwx 1 root root 15 Aug  3 13:51 NCDrive_0 -> /media/USBdrive

This is exactly what I want. I will now try moving the data and DB directory.

I think I understand the symbolic linking for the naming. Just not sure why I ended up with 3 'Directories" is that the correct word on the drive after partitioning.