nextcloud/nextcloudpi #69: Samba server not starting on reboot

defaultuser8, 2017-07-27:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -100 /var/log/mysql/*.log`
```
</details>

defaultuser8, 2017-07-27:
Latest build of nextcloudpi (Fresh)
Booting from usb harddrive
Samba server not starting on reboot
using this command to start samba server "sudo /etc/init.d/samba restart"

nachoparker, 2017-07-28:
I just tested on a fresh image

 - in `nextcloudpi-config` activate `samba` ( `ACTIVE=yes` )
 - reboot
 - check status
```
# systemctl status smbd           
● smbd.service - LSB: start Samba SMB/CIFS daemon (smbd)   
   Loaded: loaded (/etc/init.d/smbd)                       
   Active: active (running) since Fri 2017-07-28 07:06:25 UTC; 2min 58s ago                                           
  Process: 1032 ExecStart=/etc/init.d/smbd start (code=exited, status=0/SUCCESS)                                      
   CGroup: /system.slice/smbd.service                      
           ├─1088 /usr/sbin/smbd -D                        
           └─1094 /usr/sbin/smbd -D                        

Jul 28 07:06:25 raspberrypi smbd[1032]: Starting SMB/CIFS daemon: smbd.                                               
Jul 28 07:06:25 raspberrypi systemd[1]: Started LSB: start Samba SMB/CIFS daemon (smbd).
```

What modifications have you made? How did you set up the system to boot from USB drive?

I'll close because I could not reproduce it on a fresh unmodified image, but we can keep talking

closed 2017-07-28 by nachoparker

defaultuser8, 2017-07-28:
First off this is how i'm booting from a usb had drive:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

As for modifications, i am adding these packages to your image through apt.
ntfs-3g
samba samba-common-bin(just to check and make sure its up to date)
rsync
update and upgrade
dist-upgrade
apt-transport-https
getting a  key and adding the plexmediaserver repository
update
plexmediaserver

In that order, and somewhere along the way its breaking the samba server from restarting on boot.

Sent using Boxer

defaultuser8, 2017-07-28:
Oh and i forgot to mention this started about 2 weeks ago while i was still booting from the sd card after an update/upgrade.  I thought since its not working correctly i'd start from scratch and switch to booting from the hard drive.
At this point i'm going to have to redo it all again because nextcloud is also not working, i am getting a malformed server configuration.  I only mentioned the samba issue because it was all working fine until i did the update.
Sent using Boxer

nachoparker, 2017-07-28:
yeah, I have not experimented with booting from the drive.

If you are going to start from scratch, it would be great if you first just move the OS to the USB drive, and see if Nextcloud works.

Then enable samba and see if it works.

It would be nice feedback.

btw: a good thing of starting from scratch is that you can use the new web interface ;)

Thanks!

defaultuser8, 2017-08-01:
finally got around to messing with my pi server, I downloaded the newest
image and "etched" it to an sd card and booted up my pi from scratch.
Nextcloud started fine, but when i added my network share to samba and
restarted i got an error.

pi@raspiNAS:~ $ sudo /etc/init.d/samba restart
[ ok ] Restarting nmbd (via systemctl): nmbd.service.
[....] Restarting smbd (via systemctl): smbd.serviceJob for smbd.service
failed. See 'systemctl status smbd.service' and 'journalctl -xn' for
details.
 failed!

So it looks like samba may have started, but when i added my share and
restarted it failed.  BTW, Have i thanked you for your work on this
project? If not Thanks, your image has been a great experience so far.  I
tried to manually install nextcloud and never could get it work ing right.
When i found your image i couldn't believe how well it worked. Anyway, back
to the issue at hand. I didn't open a new issue on GIT since we are already
hashing it out via email.

That failure was last night, so after running some errands this morning, I
downloaded the image again and flashed to my hard drive and booted from
that to see if maybe i had corrupted my download or my flash, but i got the
exact same failure.  Once again i really appreciate all your hard work, I
know how time consuming it can all be, and if there is anyway I can help I
will do my best to contribute where my meager talents benefit the most.
Sincerely,
Dax McCrory

nachoparker, 2017-08-07:
hi, what is the output of `systemctl status smbd`?

defaultuser8, 2017-08-07:
I'll have to figure out how to get that and get back to you.

Sent using Boxer

nachoparker, 2017-08-07:
In the NextCloudPi_07-21-17 image this didn't happen, but in the `NextCloudPi_07-29-17` it does.

I'll mark as duplicate of #82

defaultuser8, 2017-08-15:
Just an update, I downloaded the latest image yesterday and "etched" my hard drive and started all over from scratch.  Everything is back working and i have added hdparm and plexmediaserver without any issues.  FYI, if you ever try to run from a hard drive you will need to resize the primary partition manually because the tool that does this for you on an sd card doesn't work on a hard drive.  I followed the directions on this page :
http://elinux.org/RPi_Resize_Flash_Partitions

"Manually resizing the SD card on Raspberry Pi

You can also resize the partitions of the SD card that your Pi is running on.

First you need to change the partition table with fdisk. You need to remove the existing partition entries and then create a single new partition than takes the whole free space of the disk. This will only change the partition table, not the partitions data on disk. The start of the new partition needs to be aligned with the old partition!

Start fdisk:

sudo fdisk /dev/mmcblk0

Then delete partitions with d and create a new with n. You can view the existing table with p.

    p to see the current start of the main partition
    d, 3 to delete the swap partition
    d, 2 to delete the main partition
    n p 2 to create a new primary partition, next you need to enter the start of the old main partition and then the size (enter for complete SD card). The main partition on the Debian image from 2012-04-19 starts at 157696, but the start of your partition might be different. Check the p output!
    w write the new partition table

Now you need to reboot:

sudo shutdown -r now

After the reboot you need to resize the filesystem on the partition. The resize2fs command will resize your filesystem to the new size from the changed partition table.

sudo resize2fs /dev/mmcblk0p2

This will take a few minutes, depending on the size and speed of your SD card. "

Thank you again for all your hard work!!

Sent using Boxer

On Aug 1, 2017 12:55 PM, Dax McCrory <daxmccrory@gmail.com> wrote:

finally got around to messing with my pi server, I downloaded the newest image and "etched" it to an sd card and booted up my pi from scratch.  Nextcloud started fine, but when i added my network share to samba and restarted i got an error.

pi@raspiNAS:~ $ sudo /etc/init.d/samba restart
[ ok ] Restarting nmbd (via systemctl): nmbd.service.
[....] Restarting smbd (via systemctl): smbd.serviceJob for smbd.service failed. See 'systemctl status smbd.service' and 'journalctl -xn' for details.
 failed!

So it looks like samba may have started, but when i added my share and restarted it failed.  BTW, Have i thanked you for your work on this project? If not Thanks, your image has been a great experience so far.  I tried to manually install nextcloud and never could get it work ing right.  When i found your image i couldn't believe how well it worked. Anyway, back to the issue at hand. I didn't open a new issue on GIT since we are already hashing it out via email. 

That failure was last night, so after running some errands this morning, I downloaded the image again and flashed to my hard drive and booted from that to see if maybe i had corrupted my download or my flash, but i got the exact same failure.  Once again i really appreciate all your hard work, I know how time consuming it can all be, and if there is anyway I can help I will do my best to contribute where my meager talents benefit the most.

Sincerely,

Dax McCrory

nachoparker, 2017-08-15:
Cool! that's great info. 

Thanks for sharing

defaultuser8, 2017-09-16:
I ssh'd into my pi today and saw this:

Any idea why it says i have mail, I don't have mail setup on my pi
Thanks,
Dax McCrory

nachoparker, 2017-09-17:
Sorry, you saw what?

defaultuser8, 2017-09-17:
I'm sorry did the screenshot i attached not show up, when i sshed into my pi i got a notification the i had mail.  But as i stated i dont have email stup on my pi.  So i'm asking if that is a function of nextcloud and if so how would i find the email.  

Sent via the Samsung Galaxy Tab® 4, an AT&T 4G LTE tablet

nachoparker, 2017-09-17:
there is email settings in NCP but I have not seen what you describe. Interesting.

Could you start a new thread in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)? this is offtopic here

thanks