nextcloud/nextcloudpi #70: Allow nextcloudpi devices to seed the nextcloudpi torrent

jonim8or, 2017-07-28:
In a blogpost you mentioned the cost of hosting the nextcloudpi image. Maybe it is a good idea to use torrent more:

1. get new updates via torrent
2. (optional, can be enabled/disabled via checkbox) let the nextcloudpi instances seed the latest image for a while

This way all the updating-related traffic would be reduced to almost zero, and any torrent-download of the whole image will be blazingly fast

I know that these are actually two suggestions in one, but I post them together because I think they are related a lot.

nachoparker, 2017-07-28:
could be a little bit of work to pull off... but brilliant idea!

thanks

grphx7, 2017-08-25:
I would love to be able to seed the image in a torrent, on my pi.  Please keep in mind the throughput on a pi is somewhat slow though so maybe don't allow the torrent to have 100% of the bandwidth?  I do seed the image torrent on my main server though to help out currently.

nachoparker, 2017-08-28:
> I do seed the image torrent on my main server though to help out currently.

Thanks for that! :D

sunjam, 2017-09-08:
fyi, this could be achieved using [ocDownloader](https://apps.nextcloud.com/apps/ocdownloader) app, which is currently being maintained by an active fork on Github... see their [github page here](https://github.com/e-alfred/ocdownloader).  It would be cool to add via the web menu, much like the News App Updater.

nachoparker, 2017-09-08:
cool, thanks for the info!

nachoparker, 2017-10-06:
At this moment we don't have the resources to implement this, and also right now the torrent state is healthy. 

Close for now

closed 2017-10-06 by nachoparker

sunjam, 2017-10-06:
Any idea of what resources are needed to implement it?

nachoparker, 2017-10-06:
I mean mostly manpower, pretty busy right now

The idea is great though, thanks

 :D