nextcloud/nextcloudpi #71: Email Failure

cottoa00, 2017-07-28:
### PROBLEMS
Having problems with Email test in Additional Settings. Have dropdown set to PHP (only option) and my email address filled in as anthony.cotton @ energy-conscious.net

As a side issue, I have the Email app installed and can receive emails but am unable to send.
Perinent log errors included below.

### Version of NextCloudPi are you using? v0.17.13... just upgraded today.. have had error since install though

### Base image that you installed on the SD card? NextCloudPi_06-28-17 )

### Expected behavior... Additional Settings/Send Email... succesful email sent

### Actual behaviour... Failure with message
A problem occurred while sending the email. Please revise your settings. (Error: Email could not be sent. Check your mail server log)

### Include logs
<details>
<summary>Nextcloud logs</summary>

Error | index | TypeError: Argument  1 passed to OCA\Mail\Model\IMAPMessage::hordeToAssoc() must be an  instance of Horde_Mail_Rfc822_Address, instance of  Horde_Mail_Rfc822_Group given, called in  /var/www/nextcloud/apps/mail/lib/Model/ConvertAddresses.php on line 74 | 2017-07-28T11:18:04-0400
-- | -- | -- | --
Error | index | TypeError: Argument  1 passed to OCA\Mail\Model\IMAPMessage::hordeToAssoc() must be an  instance of Horde_Mail_Rfc822_Address, instance of  Horde_Mail_Rfc822_Group given, called in  /var/www/nextcloud/apps/mail/lib/Model/ConvertAddresses.php on line 74 | 2017-07-28T11:17:30-0400

</details>

cottoa00, 2017-07-29:
An update to this issue... my installed Email app is working for inbound mail and although outbound mail gives me a "Server Error!" within the email app... the mail is getting thru. So not sure why the Admin section is failing although in the email app I had to setup IMAP and SMTP settings and don't remember doing that when installing NCPi and did not see anything in  nextcloudpi-config.

cottoa00, 2017-07-29:
Forgot to ask... know this is a NextCloud Email app issue but might you know how to "subscribe" to folders in that app? My gmail accounts automatigically subscribed to all folders whereas accessing accounts on my hosted email service did not. My biggest concern with NextCloud in general is that there are no buttons or knobs to tweak things at the UI level. Would love a "map" of major components (Apache, Maria, Apps, etc) where they live and where their config and log files are ;-)

cottoa00, 2017-07-29:
Final update... found the location of config.php and followed the Version 12 Admin instructions to change the email settings from "php" to "smtp"... after some fussing with ownership privileges got the email to work thru my hosted SMTP server. I'm good to go. Thanks.

closed 2017-07-29 by cottoa00

nachoparker, 2017-07-29:
Good job,

I have not thought about setting the email config through `nextcloudpi-config`. I'll look into it if I have the time ;)