nextcloud/nextcloudpi #73: v0.17.19 WebPanel access

Shadowstreik, 2017-07-31:
In this latest update run from NC-Update in SSH, I fire up Firefox and punch in the https address with :4443 at the end. I'm told it cannot connect. Chrome reports the connection was refused.

What is needed to ensure I can access the web panel? Using Linux Mint.

Thank you.

nachoparker, 2017-07-31:
Are you using `https://domain:4443`? try `https://ip:4443`

JimmyKater, 2017-07-31:
https://domain:4443 isn't working for me. but https://local-ip:4443 does

nachoparker, 2017-07-31:
it's because of the HSTS security mechanism. I will improve this so we can use the domain name

nachoparker, 2017-07-31:
implemented in `v0.18.1`. 

For it to become effective, launch `letsencrypt` from `nextcloudpi-config` (or web) again

closed 2017-07-31 by nachoparker

Shadowstreik, 2017-07-31:
- Updated to v0.18.5
- Launched Firefox. Typed the local IP -> https://ip:4443
- Firefox reports it cannot connect. Problem still exists.

nachoparker, 2017-07-31:
Did you download the last image and start from scratch? 

If not, did you follow the [instructions for manual installation](https://ownyourbits.com/2017/07/24/nextcloupi-gets-a-web-interface/) of the web interface?

Shadowstreik, 2017-07-31:
I logged into SSH where it said an update was available and to type 'sudo ncp-update'. Did that and NCP was updated to v0.18.5. The web panel acces was expected to be functional after the update was performed, but wasn't in effect. Please resolve.

Shadowstreik, 2017-07-31:
Also performed a reboot for a full restart and allow anthing needing a reboot to take effect. The reboot had no effect.

nachoparker, 2017-07-31:
Please, read the linked post carefully. `sudo ncp-update` is not enough

If you come from an old image, you have to copy and paste the following.

```
sudo su
cat > /etc/apache2/sites-available/ncp.conf <<'EOF'
Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
  SSLEngine on
  SSLCertificateFile      /etc/ssl/certs/ssl-cert-snakeoil.pem
  SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key
</VirtualHost>
<Directory /var/www/ncp-web/>
  Require host localhost
  Require ip 127.0.0.1
  Require ip 192.168
  Require ip 10
</Directory>
EOF
a2ensite ncp
 
mkdir /home/www -p
chown www-data:www-data /home/www
chmod 700 /home/www
 
cat > /home/www/ncp-launcher.sh <<'EOF'
#!/bin/bash
DIR=/usr/local/etc/nextcloudpi-config.d
test -f $DIR/$1 || { echo "File not found"; exit 1; }
source /usr/local/etc/library.sh
cd $DIR
launch_script $1
EOF
chmod 700 /home/www/ncp-launcher.sh
echo "www-data ALL = NOPASSWD: /home/www/ncp-launcher.sh" >> /etc/sudoers
systemctl reload apache2
```

It is explained in the article I linked, in the 'Installation' section

Shadowstreik, 2017-07-31:
I've been looking through that. I have MS, so grasping things can often be difficult. I'm looking for the path for nexcloudpi.sh. Once I find that, putting in that code is easy.

nachoparker, 2017-07-31:
The easiest path is to backup, then install the new image, then restore nextcloud.

If you want to copy and paste the code, you have to do so in a terminal, don't need to use `nextcloudpi.sh`.

How are you accessing the RPi? through SSH, or plugging in a physical keyboard?

Shadowstreik, 2017-07-31:
I use SSH, but also run Filezilla if I have a file to move over or back and so-on.

Shadowstreik, 2017-07-31:
The image I installed was the June 28 image.

nachoparker, 2017-07-31:
ok, then just SSH and copy-paste that code ;)

nachoparker, 2017-07-31:
on the command line

Shadowstreik, 2017-07-31:
Excellent. Thank you very much. I think that may be good to include in the instructions. It's been performed and is running great. Thank you, again.

nachoparker, 2017-07-31:
Good job!

Well, if you download the latest image (the only one that shows on the webpage) you don't need to do any of this. It's only to upgrade from old images ;)

Shadowstreik, 2017-07-31:
I have the latest image, now, and will hold it aside should I every need to do a full rebuild again.

nachoparker, 2017-07-31:
cool

I still think that it's easier to backup, install the new image and restore that what you did. I recommend that you just backup and bring your instance of NextCloud to the newer image.

That being said, I try really hard not to introduce "breaking changes". Only really deep changes like the web interface would need you to start from scratch

btw: the latest image is like... 5 minutes old xD ( NextCloudPi_07-29-17 )

Shadowstreik, 2017-07-31:
I just completed downloading it. Thank you for the heads-up.