nextcloud/nextcloudpi #75: general updating with nextcloudpi

JimmyKater, 2017-07-31:
hey @nachoparker,

before i started using ncpi i ssh'ed in weekly to update my debian that was running on my pi. if i understood it correctly the new v17.x ncpi-config does that job for me if marked. right? so no need for me to check for updates (since raspbian isnt featuring security updates, only)... but never the less - i could do it though, right? without crashing the system... ;o)

and what about updating the nc-instance itself? better with nc-update or could it be done "manually" as well from the admin section of nc with the installed updater? 

as you can see - just some questions. no issues so far.

nachoparker, 2017-07-31:
> before i started using ncpi i ssh'ed in weekly to update my debian that was running on my pi. if i understood it correctly the new v17.x ncpi-config does that job for me if marked. right

If you are refering to `unattended-upgrades`, it has been around for a loong time ;)

But yes, if you activate `unattended-upgrades`, the system will update itself automatically. You can see the update logs and stuff. More info [here](https://ownyourbits.com/2017/03/21/automatic-security-updates-on-a-debian-system/)

> but never the less - i could do it though, right? without crashing the system... ;o)

You can certainly still do it manually ;)

> and what about updating the nc-instance itself? better with nc-update or could it be done "manually" as well from the admin section of nc with the installed updater?

`ncp-update` will only update NextCloudPi stuff (such as the new web interface or `nextcloudpi-config`, for updating NextCloud itself you should use the 'updater app' from the admin section

closed 2017-07-31 by nachoparker

JimmyKater, 2017-07-31:
> More info here
yapp... i was referring to that.

> ncp-update will only update NextCloudPi stuff (such as the new web interface or nextcloudpi-config, for updating NextCloud itself you should use the 'updater app' from the admin section

that answers all of my questions. thank you so much